winder/Universal-G-Code-Sender #623: GUI Scaling for high DPI/4K screen?

walkinplaces, 2017-04-16:
Are there any tricks to adjust the target resolution of the program? I have a 4k laptop monitor, and the interface comes in really, really small. I'd like to avoid changing to a lower resolution every time I need to use the program... is there workaround?

winder, 2017-04-17:
@walkinplaces are you using the classic or platform version?

walkinplaces, 2017-04-18:
@winder I am using the classic version.

winder, 2017-04-18:
@walkinplaces you can try using a Java9 early release version, it is supposed to fix this. Alternatively, I've been told the Platform version has better support for HiDPI screens.

closed 2017-04-18 by winder

winder, 2017-08-19:
Suggested on the Inventables forum: https://discuss.inventables.com/t/solved-ugs-window-too-small/40923/4
```
Windows 10 machine:

Find java.exe you installed.
Right click -> Properties
Go to Compatibility tab
Check Override high DPI scaling behavior.
Scaling performed by: choose System
```