winder/Universal-G-Code-Sender #625: Inconsistent "Return to Zero" results

cjbaar, 2017-04-19:
I am having difficulty getting useful and consistent results from "Return to Zero" function. Maybe I am not using it correctly, but I am not sure how else to accurately reposition the spindle after changing bits.

Sometimes the command just sends "G90 X0 Y0 Z0". When it does this, my arduino-based board processes the Z-axis first, so the bit ends up dragging across the work piece. Other times, it tries to raise the spindle first, but comes up with completely random and large values, such as "G90 Z12.7." My Z-axis only has about 3" of play, so it immediately shoots to the top and the motor starts grinding before I can react and hit the kill switch. I have even seen Z values sent over 4,800 inches.

I can write my own in the macros, I suppose... but is there a way to control what the "Return to Zero" function sends?

jahnj0584, 2017-04-19:
I sent this issue before. Are you on the latest version? A work around is
jogging the Z to its machine 0 first then clicking it

cjbaar, 2017-04-19:
I am using version 1.0.9, which appears to be the latest in the stable branch. I have tried manually moving the spindle close to home before clicking it, in order to prevent the bit from dragging too far. However, this doesn't help anything when UGS decides to send a "Z4896.3" command and raise the spindle 400 feet into the air.

jahnj0584, 2017-04-19:
Try with the platform version

winder, 2017-04-20:
@cjbaar those issues should be fixed in the nightly build version

closed 2017-04-20 by winder

cjbaar, 2017-04-20:
So far, much better with the platform version. Guess I'll stick with that from now on. Thanks for the input!

winder, 2017-04-20:
@cjbaar Thanks for the update!

cjbaar, 2017-04-21:
Actually, I was to quick to respond. Doing some more work this evening (using platform edition), and the return button issued:
>>> G90 G0 Z0
>>> G90 G0 X0 Y0
>>> G90 G0 Z0
Again, since the Z0 was issued first, I got a nice scratch across the top of the piece. I would just like to be able to edit what commands are sent so I know that they are consistent.
Thanks again for all your work!

joacimb, 2017-04-21:
I've experienced scratched surfaces by homing myself, mostly because my crappy home built CNC machine isn't absolutly leveled... =) 

I saw that the max position has previously been used but removed because of people experiencing problems with this feature. I'm thinking that we could instead use an optional setting for Z clearance distance to be used?

jahnj0584, 2017-04-21:
If you're scratching your Z is too low anyways.

winder, 2017-04-21:
Unfortunately, this is all expected behavior at this point.

I used to have a feature to raise the Z axis to the maximum Z-height, but there was a bug in it causing the original issue. I believe it was somehow related to homing but was never able to reproduce it and ended up disabling the feature.

Shortly after disabling the feature there were reports that people would use return-to-zero while Z-height was less than zero leading to broken tools. So I added a basic check that if the Z-height is less than 0 `G90 G0 Z0` gets sent first which prevents the broken tools but now work surfaces can be scratched.

cjbaar, 2017-04-21:
OK... so back to my original question... can we have option to put in our own commands for these buttons. I don't see any way to create macros in the platform version. I see a panel for them, but it looks like maybe they aren't implemented yet? But even it were, that puts things on another tab, and now I have to remember what "M1" "M2" "M3" etc mean.

I am in the process of adding limit switches on my setup. I know many machines do not have them, but once installed I am sure a modified version of the original z-lift could be implemented more safely.

winder, 2017-04-21:
@cjbaar macros are the only configurable buttons, in the platform, you can configure them from the preferences (sounds like that menu isn't visible enough). Macros in the nightly build are nameable, so no need to remember "M1" "M2" "M3", etc.

cjbaar, 2017-04-21:
OK, thanks. You are right. I did not see them looking through the preferences the first time.

cjbaar, 2017-04-23:
Thanks again for the info, @winder. I can see where the macros are created (and named), but when I click on one of the buttons in the Macros panel, it doesn't send any commands. Is there another step to getting these enabled?

vMeph, 2017-04-23:
it should be working fine, 
you got to name and set g code parameters
exemples:

Macro name              G code                            Macro Description

Spindle ON               S1000 M3                        turns on spindle
Spindle OFF               S0 M5                             turns off spindle
G54                           G10 P1 L20 X0 Y0 Z0       select work offset

cjbaar, 2017-04-25:
Thanks, @vMeph. I have setup some macros, as shown here. However, when I click on the buttons in the macros panel, they don't do anything. No commands are sent. (Not sure if I should continue this here, or post a separate issue.)
[image]