winder/Universal-G-Code-Sender #626: Is "Folder Mode" a possibility?

bashkan, 2017-04-19:
Hi,
I am not sure if this is the right place to post it, but I am new to this environment so please forgive me if I am doing something wrong.
I believe Universal G Code Sender is a great software and it is the perfect choice for all the CNC people out there. However, I need a feature for my setup which is somehow different than a classical CNC. My device is a 2 axis machine which draws on sand by moving a steel ball (inspired by this: https://www.kickstarter.com/projects/1199521315/sisyphus-the-kinetic-art-table ) . I know how to cleanup the Gcode files for 2 dimensional movement but I need a software which will:
-work on Raspbian (RaspberryPi is controlling the Grbl Arduino)
-start when RaspberryPi boots 
-look for a folder full of gcode files (at least 100 gcode files)
-send gcode files to Grbl Shield one by one in an infinite loop (when the last file in the folder is sent, it will start all over again)
I have researched customizing the Universal Gcode Sender source code but I have no experience in Java/Net Beans/Maven etc. I have even looked at the workflow plugin example which does something similar to what I need but it is way beyond my programming skills. (I have some experience in Pascal/Object Pascal but getting it even work on RPi is a nightmare, let alone developing/debugging/compiling on it)
So, back to the title of the post, how hard is it to make a tab (like "Commands", "File Mode" etc) named "Folder Mode" which achieves the results I mentioned above? Or is there any other solution that you know that can point me to the right direction?
Again, I do apologize if this is not the right place. 
Thanks in advance.
Koray Ozbay

winder, 2017-04-19:
@bashkan I'm not planning to implement anything like this, but you should be able to make something to accomplish your specific task. Take a look at the grbl stream.py or simple_stream.py programs:
https://github.com/grbl/grbl/blob/master/doc/script/stream.py
https://github.com/grbl/grbl/blob/master/doc/script/simple_stream.py

To start with you could wrap one of those with another script that iterates your directory and streams each program one after another:
```
import os
import time

directory = "/folder/with/gcode/files"
stream_script = "/location/of/stream.py"
device = "/dev/tty.yourArduino""

while (true):
    for filename in os.listdir(directory):
        os.system("python %s %s %s" % (stream_script, os.path.join(directory, filename), device))
        time.sleep(5)
```

bashkan, 2017-04-19:
@winder Thank you very much for your prompt reply. I will check the links that you sent right now.

closed 2017-04-20 by winder