winder/Universal-G-Code-Sender #628: Raspberry pi - Platform - Netbeans stops after "reading module storage"

strider460, 2017-04-20:
I've been out of the UGS / GRBL world for a while now and I am just getting around to updating both the arduino and raspberry pi for my cnc, since I am really excited about the feeds and speeds override in grbl 1.2 and the modularity of the new UGS platform layout, but I am having some trouble. At first I thought it was a Java problem so I updated to the newest raspbain on my Pi and it's running java 1.8.0_65 which I believe is 8 (?) and I made the ugsplatform shell file in the bin folder executable, but when the splash screen for netbeans comes up the progress bar stops when the status indicator says "reading module storage" and closes out. It is a Raspberry pi 3 so it shouldn't be too outdated. 

I know the "how to run" section is on the To-Do list but I was just hoping it was something simple I was missing? any ideas?

winder, 2017-04-20:
If you run that ugsplatform shell script from the terminal are there any error messages in the output?

strider460, 2017-04-20:
I guess it is worth mentioning i have the "ugsplatform" folder, downloaded from the new UGS websites download page, and i am using it from there. 

The output of the terminal is fairly lengthy; the whole thing is in a text file in the link below:
[https://drive.google.com/open?id=0B2GD-Zc8XyOHaDlydl9HODdtSDg](url)

it seems to be a loop of the below error message in a couple of variations (error opening, cannot opening, ect) and the retry count goes  2, 3, 4, -1, 2, 3... and so on. 

Apr 20, 2017 1:45:13 PM org.netbeans.JarClassLoader$JarSource$1 call
WARNING: Error opening /home/pi/Desktop/ugsplatform/ugsplatform/core/locale/._core_ugsplatform.jar (exists=true) retry: 3
java.util.zip.ZipException: error in opening zip file
	at java.util.zip.ZipFile.open(Native Method)
	at java.util.zip.ZipFile.<init>(ZipFile.java:219)
	at java.util.zip.ZipFile.<init>(ZipFile.java:149)
	at java.util.jar.JarFile.<init>(JarFile.java:166)
	at java.util.jar.JarFile.<init>(JarFile.java:145)
	at org.netbeans.JarClassLoader$JarSource$1.call(JarClassLoader.java:546)
	at org.netbeans.JarClassLoader$JarSource$1.call(JarClassLoader.java:537)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at org.netbeans.JarClassLoader$JarSource.getJarFile(JarClassLoader.java:572)
	at org.netbeans.JarClassLoader$JarSource.listCoveredPackages(JarClassLoader.java:651)
	at org.netbeans.JarClassLoader.getCoveredPackages(JarClassLoader.java:983)
	at org.netbeans.JarClassLoader.<init>(JarClassLoader.java:176)
	at org.netbeans.JarClassLoader.<init>(JarClassLoader.java:153)
	at org.netbeans.MainImpl$BootClassLoader.<init>(MainImpl.java:272)
	at org.netbeans.MainImpl.execute(MainImpl.java:178)
	at org.netbeans.MainImpl.main(MainImpl.java:85)
	at org.netbeans.Main.main(Main.java:83)

strider460, 2017-04-20:
Clicking on the link to my drive file doesn't work, I think that was my bad. Copy pasting it looks like it works though.

winder, 2017-04-20:
@strider460 my only guess is that one of the files is corrupt and the error is not being handled very well, could you try re-downloading the archive?

strider460, 2017-04-20:
wow... replaced the folder with a fresh download and it worked like a charm. I don't know where I got the file from but it was more than likely something I did.  

Thanks @winder !

closed 2017-04-20 by strider460