winder/Universal-G-Code-Sender #629: Firmware Version

mahid1, 2017-04-20:
How can i stop displaying grbl firmware version string?

winder, 2017-04-20:
Do you want to prevent the version string from being displayed in the UGS console?

Remove this line: https://github.com/winder/Universal-G-Code-Sender/blob/717da06f8d7c724b7212daa4b6a42a365c89ebc9/ugs-core/src/com/willwinder/universalgcodesender/GrblController.java#L157

mahid1, 2017-04-21:
Thanks buddy, 
Do i need to have a specific compiler to do this? Also please let me know if there is a way i can add lcd with grbl. I am using atmega2650.

mahid1, 2017-04-21:
Unable to execute UGS. 
Virus definition detected after execution of sim

winder, 2017-04-21:
Any Java IDE should work, you can try the documentation here: http://winder.github.io/ugs_website/dev/getting_started/

closed 2017-05-01 by winder