winder/Universal-G-Code-Sender #634: GRBL NOT FINISHED LOADING, HELP!

gdc5414, 2017-04-27:
[image]

So Im using Windows 10, GRBL.9, Arduino, and UGS. I had this working a few days ago but the z-axis stopped working. So I got another shield and remounted it. Now I can't connect. It says in UGS "**** Connected to COM4 @ 115200 baud **** But then thats it. If I do anything, it gives me a random code, or the one pictured.

I have updated drivers, removed drivers, deleted drivers and redownloaded, everything. Redownloaded UGS, even found a tutorial and followed it step by step. Still can't connect. Tried another Arduino board. Same issues. What is going on. Someone please help!

winder, 2017-04-27:
This usually means you're using the wrong baud rate or grbl isn't flashed properly.

gdc5414, 2017-04-27:
I tried ever baud rate.

Just for giggles, what is the correct procedure on flashing it?

winder, 2017-04-27:
The documentation here, it depends on your platform: https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino

rmisko11, 2017-04-27:
Don't know the correct procedure but this works for me:

avrdude -p ATmega328P -P /dev/ttyUSB0 -c arduino -b 57600 -U
flash:w:grbl.hex:i

I think I had to use '57600' to correct the problem you see, but I am not
sure as I forgot to note my fix for the same problem.

rmisko11, 2017-04-27:
Recently, I have a problem where the arduino won't even start after
flashing but that is because I am trying to flash more than the available
flash (32K - boot loader space).  I tried enabling coreXY in grbl 1.1f but
I can't get it to fit in an Arduino Uno.

closed 2017-05-02 by winder