winder/Universal-G-Code-Sender #636: Unable to use macros

cjbaar, 2017-05-01:
In the absence of being able to customize the stock buttons (return to zero, $h, etc)... I am trying to create macros for precise control. However, I can't get them to work. I have created the information in the settings, as shown below, but clicking on the button in the "Macros" panel does not do anything, and no commands are sent. Is there something else I need to enable? Thanks.
[image]

closed 2017-05-01 by winder

cjbaar, 2017-05-13:
I see this issue was closed quickly, but I don't see any action taken it. The latest version of platform (downloaded last night) still does not have these fixed. I am unable to use any macros at all

vMeph, 2017-05-13:
Macros works dont know what you trying to achive home is issue by comand $H if you have switches .
your home macro is G91 X259 Y50 ,it seems you work on positive space usualy on cnc you work on negative side.
You have machine position and work positions so lets say you are at home X0 Y0 Z0 and the piece you working on is at X-100 Y-50 Z-7, you jog to there and then you set your working position by using the command G10 P1 L20 X0 Y0 Z0,That will set your work coords to  X0 Y0 Z0 ,WCO on G54,so now you have your machine coords at X-100 Y-50 Z-7 and work coords at X0 Y0 Z0,so when you have a gcode file and you set there G54 and you at home ,when you start file it always will go to the G54 start point WCO X0 Y0 Z0

cjbaar, 2017-05-14:
The content of the macros is not what is relevant here; I just entered random commands I was using at the time. The macros do nothing. Clicking on the button sends no commands. It doesn't matter what the commands are set to.

winder, 2017-05-14:
@cjbaar there was a bug where the buttons in the macro settings weren't clickable.  It was fixed which is why the issue was closed. You can see the code change referenced right above your comment yesterday.

The most common way to use macros is from the macro window in the main application and this has always worked. If you closed this window you can re-open it from window menu. Both methods should now be working. Are you running a recent nightly build?

[image]

cjbaar, 2017-05-14:
I am not trying to run the macros from the settings pane; I am clicking on them in the "Macros" panel of the main application window. I just re-downloaded yesterday, using the "UGS Platform" link under the "2.0 Nightly builds" on the main github page. Is there another link I should be using?

cjbaar, 2017-05-14:
I feel like there is some confusion as to what my issue is. Here is how I am trying to use the macros functionality. I have created a macro in the settings.

[image]

Then, I close the settings. The new macro button show up in the macros pane, in the upper-left corner of the GUI window.

[image]

Then, I click on this button. It is my understanding that clicking on the button should send whatever command was created in the preferences. However, I get nothing. No command is sent; no data is shown in the console output.

[image]