winder/Universal-G-Code-Sender #640: UGS multiplies all GCode feedrate by 2.6

jiarko, 2017-05-01:
Hi,

I've noticed the UGS multiplies all feedrates by a factor of 2.6. This happens for both "classic GUI" and new "UGS Platform".
here n example:
my GCode -->G2 Y-233.072 Z2.5 J-0.4 F960

UGS table --> G2Y-233.072Z2.5J-0.4F2496

that weird thing is that in the command table I see my correct code in the second column and the "multiplied" one in the first column, so I guess it shouldn't be a parse issue.. this happens for ALL feedrate command

can someone help me?

jahnj0584, 2017-05-01:
Sounds like you're missing units. A standard to metric would be a 2.54
ratio

jiarko, 2017-05-01:
That's what I've thought... but than the dimensions would have been wrong not the feedrate... this is the beginning of my GCode...

(T1  D=4 CR=0 - ZMIN=-16 - flat end mill)
G90 G94
G17
G21
M9
T1 M6
S10000 M3
G54

winder, 2017-05-01:
@jiarko it sounds like you accidentally enabled the feed rate override setting
[image]

jiarko, 2017-05-01:
you're right... I didn't know that there is this panel, so maybe some accidentally hit shortcut enabled this...

I could check this only because you send me the screenshot because in my version there is no labels beside the checkbox... as in the snapshot below

[image]

BTW, when I start the program I get an empty window with an error icon, do you know how to fix these issues?

[image]

thanks again

winder, 2017-05-01:
@jiarko Your language isn't fully localized. I thought those messages were supposed to fall back to english. If you'd like to help finish the localization UGS is using a service here: https://poeditor.com/join/project/2J2hB5I41Z

jiarko, 2017-05-01:
do you think this is the cause of not having labels beside the "Controller options" window? cause I didn't choose any language setup, in fact all descriptions are in english...

thanx for your help

winder, 2017-05-01:
The "sender options" tab in the screenshot above has a language selector at the bottom, what is it set to?

jiarko, 2017-05-01:
your are (again) right. There was Italian as set language, when set English and restart application now I can see the labels.

thanx again!!!

closed 2017-05-01 by jiarko

jahnj0584, 2017-05-01:
True. Check your CAM, and the feed over ride (but that mazes at 200% I
think)