winder/Universal-G-Code-Sender #641: mvn command not found ( fedora 25 )

J-Dunn, 2017-05-01:
> Universal Gcode Sender is a self-contained Java application which includes all external dependencies, that means if you have the Java Runtime Environment setup UGS provides the rest. 

run_classic.sh  contains:
```

mvn install
mvn exec:java -Dexec.mainClass="com.willwinder.universalgcodesender.MainWindow" -pl ugs-core

```

to which I get:

> mvn : command not found 

So do I need to install maven?  "Java project management and project comprehension tool" ? Huh?

winder, 2017-05-01:
You found one of the scripts I use during development, you can download the packaged application here: http://winder.github.io/ugs_website/download/

closed 2017-05-01 by winder

J-Dunn, 2017-05-01:
I got the link from the Shapeoko website list of viewers. 
https://www.shapeoko.com/wiki/index.php/Previewing_G-Code
They show which does not look like a dev. version. 
https://github.com/winder/Universal-G-Code-Sender

If they are providing a bad link may be you could tell them where they should be pointing people.

Thanks for the rapid response.