winder/Universal-G-Code-Sender #644: Hai GRBL version 0.8c

jnivard, 2017-05-02:
Hai is it posible to use the older GRBL version 0.8c in combination withe the UGS version 1.0.9?

When I connect the CNC shield it start ok 
Grbl 0.8.c [etc]
When I print the settings, than I see that the default values are not correct and also I see that the numbering of the variables are different
When I start sending commands it is not working and I get alarms !

winder, 2017-05-02:
@jnivard it should work. What is wrong specifically? What do you see with the settings?

jnivard, 2017-05-02:
I open the connection and have the correct reply. But I have to unlock. Resetting the board is not helping When I issue a command X+ the boiard unlocked but is direct issuing the message " To unlock" 

**** Connected to COM5 @ 9600 baud ****

$Grbl 0.8c ['$' for help]
['$H'|'$X' to unlock]
<P

Grbl 0.8c ['$' for help]
['$H'|'$X' to unlock]
<P$
>>> $X
[Caution: Unlocked]
ok
<R
['$H'|'$X' to unlock]
<P
etc
USG is sendfing the <P message and the " to unlock" message every secunde. When I issue the $X message I get the ok and followed by a <R message followed by the " To unlock message" and after that it start sending the <P followed by the "To unlock" message

To simplyfy the problem I did connect only the X axis and not limit switches, but it is not working.

Thanks

vMeph, 2017-05-02:
Sounds like you have the hard limit switches enable,and sense you dont have switches will give  you alarm i would sugest disable hard limits to see whats happen them, by the way why use version 0.8 grbl and not move to the latest 1.1 grbl version?

jnivard, 2017-05-02:
No I have them off, but I think I have problems with the drivers.

closed 2017-05-02 by jnivard