winder/Universal-G-Code-Sender #648: Add way to alter feedrates and spindle speed during milling

Harvie, 2017-05-03:
Hello,
i've run into the situation where i've improperly set feedrates in my g-code and it was too late to fix that when i noticed... I could have stoped everything, jog to zero, modify g-code and run it again, but it was too much hassle... i think it would be nice if there was option to tune the feedrates online during g-code execution.

So i can eg. set feedrates to 50% or 500% while running the g-code. Of course there has to be some safety confirmation feature that will let us review that we didn't entered some ridiculous extreme value.

vMeph, 2017-05-03:
Thats already implanted on UGS

jahnj0584, 2017-05-03:
Update your software and flash GRBL 1.1

winder, 2017-05-03:
I'm not planning to add this anytime soon now that it's supported natively in grbl v1.1.

closed 2017-05-03 by winder

Harvie, 2017-05-04:
i have grbl v1.1e. it might have this feature. but there's no user friendly GUI for it in ugs-platform.

Harvie, 2017-05-04:
oh sorry. it's already there, but it's hidden by default...