winder/Universal-G-Code-Sender #649: Better keyboard jogging

Harvie, 2017-05-03:
Would it be possible to have some more responsive jogging using keyboard? So it will jog while i am holding the key and stop immediately when i release it? Right now it sends the rapid commands until i release the key, which means it may continue jogging even after the key was released.
Also it's currently not possible to jog diagonaly eg. by pressing right and forward keys at the same time. that would be usefull too.

I know it makes sense when you need to jog to exact repeatable possition. But when you are just eyeballing it makes more sense to stop immediately after key release.

BTW when i set jog distance and then i jog using keyboard it's not honored. That could lead to some collisions. Eg. when you lower distance from 20cm to 0.1mm and try to jog precisely. it jogs the 20cm unless you jog by clicking on UI button first. then you can continue to jog by keyboard in set increments.

jahnj0584, 2017-05-03:
Set the jog size to when it acts as if it's real time. For mine it was 2mm.

That issue was fixed already. Upgrade your software

winder, 2017-05-03:
I might add a special jog window for some better real-time responsive jogging: [described here](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging#joystick-implementation). The current jogging behavior probably isn't going to change very much though.

What version of UGS are you using?

vMeph, 2017-05-03:
@winder that would rox to improve the jog response

Harvie, 2017-05-04:
@jahnj0584 it's not fixed when i have to fiddle with size. and that also mean i can't rely on it.

i've noticed that when i start jogging by 100cm increments i can stop jogging imediately during process by stop button. so it can be done. just stop immediately as keys are released.

winder, 2017-10-13:
superseded by #779

closed 2017-10-13 by winder