winder/Universal-G-Code-Sender #650: G-code is in inches, but the CNC moves in mm

rlwoodjr, 2017-05-05:
I have run several g-code files in inches. Now these same g-code files convert to mm when loaded into UGS Platform. I have remove UGS and reinstalled it and it does not fix the issue. I can no longer CNC in inches

These files run fine with another g-code sender.

vMeph, 2017-05-05:
Did you see your grbl config $13=? If set to 0 is in mm if set to 1 is for inch

chamnit, 2017-05-05:
@vMeph : The $13 Grbl setting only determines the units Grbl sends back to the GUI for status reports (position, offsets, rate) and stored $# parameters. It does not effect how a gcode program is run. 

Only G20 and G21 alter how the g-code program is interpreted in inches or mm, respectively. These must be set at the start of the g-code program, every time. UGS should refer to G20/21 modal state to determine the units to send to Grbl.

@rlwoodjr : Grbl always defaults to G21 mm mode. So, after a power-cycle or reset, it's set back to G21. You can alter this default by programming a startup line by: `$N0=G20`. This will automatically run and set G20 after a reset or power cycle. 

Just beware that `$N` startup lines do not run after unlocking Grbl's alarm mode via `$X`. This is intentional to prevent startup lines from running after exiting an alarm. Primarily to prevent it from executing a motion if a user has one programmed in a startup line. Just soft-reset Grbl after an `$X` to get the startup lines to run. I know it's not ideal, but I intend to fix that soon.

winder, 2017-05-05:
@rlwoodjr Is this a hand made gcode program? You should probably put G20/G21 at the top of your program.

Older versions of UGS would modify the inch/mm state, but now only manually entered G20/G21 commands should do that. The same is true with G90/G91 absolute/relative positioning.

winder, 2017-05-05:
@rlwoodjr in case it is a UGS bug, could you share the gcode program for me to test with?

rlwoodjr, 2017-05-05:
Customers are reporting that the G20 is at the top of the file. I ask and received the file and it included the G20. It works fine on my E3. They also are reporting that they use another g-code sender and the file routes in inches.

rlwoodjr, 2017-05-05:
This is the file. I believe it is made with F-engrave.
[NO.txt](https://github.com/winder/Universal-G-Code-Sender/files/980416/NO.txt)

winder, 2017-05-06:
@rlwoodjr The only way I know this could happen is if the user somehow had G20 blacklisted by UGS.  But I'm not sure how that could happen to one person, let alone several. In that case the visualizer would also show the gcode as mm (NO.txt would be 1.97mm x 0.45mm), do you know if that is happening?

It would show up in the options with `G20` near the bottom instead of `BadLine`:

[image]

rlwoodjr, 2017-05-06:
Good call! This has worked for one of my customers. 

Yes the customers report the scale changes on the visualizer.

rmisko11, 2017-05-06:
I just looked at the file NO.txt using the nightly Platform and although it
will only show mm in the visualizer it appeared to print out, from the
Platform to a EleksDraw, correctly.

So, for me if I understand correctly, it is just a labelling issue in the
visualizer as the mm readings are equivalent to the inches in the NO.txt
file.

breiler, 2018-08-07:
The units in the visualiser are now picked up from the preferred units setting. This might be more intuitive than changing G20 and G21 or the $13 setting.

breiler, 2019-10-23:
I've tried to read through the issue a couple of times and as I understand it, this should be resolved?

UGS has a preferred unit setting now and will convert any feed rates, dro units or the model in the visualizer accordingly. Is it ok to close?

closed 2020-01-07 by breiler

ecncshopInd, 2020-09-21:
I have a CNC machine , with X axis steps/mm == 10; 
Which means steps/inch = 254;

Now say I want to move 1 inch; 

Should I set $100 = 254 and send G20 G01 X1 to GRBL

OR

Should I set $100 = 10 and send G21 G01 X25.4 to GRBL ?

Which is better ?
What does UGS does ?

ecncshopInd, 2020-09-22:
I have a CNC machine , with X axis steps/mm == 10;
Which means steps/inch = 254;

Now say I want to move 1 inch;

Should I set option1) $100 = 254 and send G20 G01 X1 to GRBL

OR

Should I set option2) $100 = 10 and send G21 G01 X25.4 to GRBL ?

OR

option3) $100 = 10 and send G20 G01 X1 to GRBL ?

Which is better ? I have already implemented option1. Is it ok to send stepsPerInch to $100. ( Iam using $13=1 along with option1)

breiler, 2020-09-22:
@ecncshopInd I would go with option 2 or 3. The setting $100 should be given in steps/mm. This will make it possible for GRBL to convert any program sent to it, metric or imperial. 

In your Gcode programming, if you prefer to work in inches use G20 otherwise G21 for metric.

ecncshopInd, 2020-09-23:
> @vMeph : The $13 Grbl setting only determines the units Grbl sends back to the GUI for status reports (position, offsets, rate) and stored $# parameters. It does not effect how a gcode program is run.
> 
referring to this Chamnit reply : What does "stored $# params" means ? is it the $100, $101 etc. settings stored in EEPROM ? But when I tested, $13=1 has no effect on the values reported by grbl to "$$" command. 

Also : Chamnit wrote : (position, offsets, rate) : In this : "rate" means Feedrate reported by GRBL in response to "?" cmd ?