winder/Universal-G-Code-Sender #653: Endless issues

gdc5414, 2017-05-08:
Windows 10, ugs 1.0.9 and arduino uno

Maybe someone can help here....
Basically I can't win with this thing. Last week my arduino stopped responding. Turns out it was a bad board. Got a new one....
Then I get the thing talking on ugs 2.0, jogs fine, can't seem to figure out homing but that's not my main issue. When I import a file from f-carve, I go to send it to the arduino and it will give me a bunch of invalid errors. So I tried doing the same thing with ugs v1.0.9 and it "sends" but dosnt so anytjing. In ugs 1.0.9, i can't jog at all. Just gives me alarm 2 or error 9. And there is 0 documentation about what these mean. I'm ready to just throw the thing in the trash. Any help?

winder, 2017-05-08:
@gdc5414 since you're only seeing error codes, I'm guessing you're using grbl v1.1?

It looks like your soft limits aren't configured properly causing grbl to reach an alarm state as soon as you move the machine. I'm not sure where the error/alarm codes are normally referenced, but I like to look them up in these tables:
https://github.com/gnea/grbl/tree/master/doc/csv

Combined with that, there's an issue (#601) with UGS right now preventing you from recovering from a soft alarm properly, so I can understand the frustration. I don't have soft alarms or homing setup on my machine but if you could share your `$$` settings I'm sure someone who does can help.

vMeph, 2017-05-08:
If you using soft limits and your file coords is bigger then your machine size settings, the machone will send a alarm rigth away and will not move.
Give us more info, grbl version, $$ your settings, shield you using.

gdc5414, 2017-05-08:
Thank you. Those codes are gold. I'm hoping these help a lot.

In the tutorial, 20 seconds after he hooks it up he homes the machine and it hits the limit switches and homes itself. When I do the same, it dosnt move. Just makes a beeping sound?

https://github.com/gnea/grbl/tree/master/doc/csv

cjbaar, 2017-05-18:
Not sure where the beeping is coming from... maybe something specific to your shield? Once you enable homing and soft limits, you won't be able to jog the spindle or send files until the homing sequence has been run any time the board is powered on or re-connected. grbl will not move unless it knows it is in a "safe" starting position. I had to tweak several settings to get my homing to work correctly. In particular, look at the masks for $5 (limit switches NO vs NC) and $23 (homing direction). If the sequence doesn't start correctly, figuring out which one of those needs adjustment can be tricky. (My board has nice LEDs that tell me when the limit switches are active, which really helps with troubleshooting.) That being said, as long as the motors are enabled, starting the homing sequence should at least move the Z axis in one direction or the other.

closed 2017-07-09 by winder