winder/Universal-G-Code-Sender #654: vISUL

vMeph, 2017-05-08:


vMeph, 2017-05-08:
@winder visualizer when you bring work piece in dimensions mm isn't same has the original piece.
is there somewhere to adjust ?
think i seen this on some post old not sure where ?

winder, 2017-05-08:
@vMeph do you mean if you load a project that uses inches, it is displayed in mm?  The dimensions are the same but they have been converted to mm.

jahnj0584, 2017-05-08:
Each dimension will be larger by the bit diameter

vMeph, 2017-05-08:
Load a project in mm and display in mm, i think @jahnj0584 answerd the question, is there anyway visualizer can read from the file the tools diameters and visualizer dimention =dimention - tool

winder, 2017-05-08:
I don't think so. For instance, if you were making a pocket how could you know from just the gcode file?

vMeph, 2017-05-08:
ya difficult to know..
i dont know how other  cam software do but for example fusion 360 on g code you have tool info independently of whatever tool path you use
ex:
(1001)
(T1  D=2.1 CR=0 - ZMIN=-6 - flat end mill)
G90 G94
G17
G21
G28 G91 Z0
G90

jahnj0584, 2017-05-09:
That's just for the machine operator. No control software applies and
offset for the bit in the visualizer. It's more for showing and helping
with clamp and fixture clearance then showing you what size the finished
part will be...

vMeph, 2017-05-09:
Ya

closed 2017-05-09 by vMeph