winder/Universal-G-Code-Sender #658: UGS platform on Win10 macros inoperative

RedHart66, 2017-05-16:
UGS platform on Windows 10 laptop PC does not enable my macros or Common Actions such as $X.
UGS works OK with my GRBL Openbuilds CNC- I have been using this a long time and wanted to try Platform.
Platform connects OK and can send commands from console but Macros and Common Actions greyed out.
Macros seem to be defined OK.
Tried $X from console first no difference.
Latest Nightly Build tried (16th. May 2017)

Grateful for any assistance, please

winder, 2017-05-25:
@RedHart66 what version of grbl are you using?

RedHart66, 2017-05-26:
@winder Grbl 0.9j other info:
Netbeans 8.1
Java version 8 update 131 build 1.8.0_131-611
Windows 10 Home
Intel Core i7-7700HQ 2.8GHZ
64 bit OS
I hope you can suggest something, thanks for the interest.

winder, 2017-05-26:
@RedHart66 I think this is the same issue as #606, in addition to the macros have you noticed any other issues?

I'm going to fix it when I can, but for now, you might consider upgrading to GRBL 1.1.

RedHart66, 2017-05-27:
@winder I have not noticed any other issues largely because I need the macros and so go back to using UGS which is working fine. If I upgrade grbl I shall report the result. Thanks for your great work on these products.

breiler, 2018-03-24:
I thought I managed to reproduce the error, but when trying the same thing several times I couldn't provoke it anymore... 

I've installed GRBL 0.9j with the latest version of UGS Platform and it seems that the problem has disappeared. 

It is possible that the problem may still occur if there where commands still queued in the BufferedCommunicator. With the fixes in #967 the buffers should be cleared when canceling, so it's less likely that this problem should occur. 

I think you can close this Will.

winder, 2018-03-24:
Thanks @breiler

closed 2018-03-24 by winder