winder/Universal-G-Code-Sender #661: usgplatform and grbl 1.1

CNCQuilter, 2017-05-17:
Flashed grbl 1.1 to Arduino using Windows 10 machine, seems to work okay under usgplatfrom.

Switched the Arduino to Linux Ubuntu box; would not operate with JOG commands.

Also when trying to see Machine Firmware Settings- Error message  grbl has not finished loading.
Switched back and forth between Windows and Linux several times, same results.

winder, 2017-05-17:
The "not finished loading" error usually means you're using the wrong baud rate or the wrong port.

CNCQuilter, 2017-05-17:
Will,

Thanks for the reply.

First of all, what is the proper etiquette to ask questions;

via github or is there a forum or just use this email?

On the Linux box, using the Arduino IDE,

the port is /dev/ttyACM0 (Arduino/Genuino uno) baud 9600

and I am able to upload sketches and they executive correctly.

With ugsplatform, the same port and baud rate are displayed.

/dev/ttyACM and 9600

The Connect Button seems to Connect and Disconnect,
in fact, when Disconnected the Machine Firmware Setting message =
Not Connected; when connected message = grbl has not finished booting.

Note: I have to execute from the command line with sudo in order to gain permission to access the port. I am a newbie to Linux so things are doubly complicated.

Dave K

CNCQuilter, 2017-05-23:
Will,

Setting baud to 115200 fixed the problem.

Two More Questions (UGSPlatform on Ubuntu).

1.G-Code Editor

Tool Menu - Plug-Ins Tab - Installed Tab, shows 12 installed including UGS G-Code Editor and it is checked as Active; but I cannot find it anywhere. Where should I look? In fact, I don't see any upload files at all, although the trace shows in Visualizer and executes correctly

2. Final Output

For my particular set-up, for G21 G91 G1 X1 results in exactly a 45 mm trace. Is there any setting(s) that translate the G1 X & Y values to corresponding reality. It would help prevent errors in my G-Code commands. For example G1 X25 would result in a 25 mm movement.

TIA,

Dave K

________________________________
From: Kender, David M.
Sent: Wednesday, May 17, 2017 3:18 PM
To: winder/Universal-G-Code-Sender
Subject: Re: [winder/Universal-G-Code-Sender] usgplatform and grbl 1.1 (#661)

Will,

Thanks for the reply.

First of all, what is the proper etiquette to ask questions;

via github or is there a forum or just use this email?

On the Linux box, using the Arduino IDE,

the port is /dev/ttyACM0 (Arduino/Genuino uno) baud 9600

and I am able to upload sketches and they executive correctly.

With ugsplatform, the same port and baud rate are displayed.

/dev/ttyACM and 9600

The Connect Button seems to Connect and Disconnect,
in fact, when Disconnected the Machine Firmware Setting message =
Not Connected; when connected message = grbl has not finished booting.

Note: I have to execute from the command line with sudo in order to gain permission to access the port. I am a newbie to Linux so things are doubly complicated.

Dave K

winder, 2017-05-25:
1. You can see the editor by selecting "edit" in the file menu while you have a gcode file loaded. This feature is still in its infancy.

2. `G21 G91 G1 X1` should move your machine 1mm in the X+ direction. If you see something else make sure your grbl settings are correct.

closed 2017-06-14 by winder

CNCQuilter, 2017-08-26:
Will - Hello again.

I am using your Universal G-code Sender Platform along with .grbl firmware in three separate and distinct CNC applications, each with their own dedicated Arduino UNO R3 and Protoneer GRBL Compatible CNC Shield.

I am working on another project (non-CNC) which uses three-axis stepper motors. The smooth integration of .grbl directed movements would be ideal for the project. However the commands would be event driven rather than canned motion instructions. Are you aware of any instances of .grbl being used in such a manner? If not .grbl, can you recommend another Arduino Stepper Library routine that smoothly and efficiently integrates three-axis stepper motor motion; or a way to implement such control using USGPlatform? Since my skills are more along the lines of integration rather than programming, it is far easier for me to adapt the creative work of other’s into my projects.

Thank you,
Dave Kender

Closed #661<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_winder_Universal-2DG-2DCode-2DSender_issues_661&d=DwMCaQ&c=3buyMx9JlH1z22L_G5pM28wz_Ru6WjhVHwo-vpeS0Gk&r=zO1BysYTt4UFmSFEv9o0SbJim_z9Tb-G4UxWKuM9N7I&m=Qih64xtSyrCTvL15DHFFnSVaHWut2ADTv4o5_LcU_z8&s=YmFliEunEKeLhI5sZEfEJdPe8Kz9kEdX7XNLkfORPxg&e=>.