winder/Universal-G-Code-Sender #662: Motor Lock

Booysena, 2017-05-18:
Hello - 
firstly, thanks for UGS - I'm enjoying your creation :-) I do have a question that I'm struggling to solve. perhaps somebody here can assist.

my cnc consists of 3 x tt6600 stepper drivers, Nema 23 and nema17 (z).
1) when I set my bit to "0" on bottom left of my work area, the I can see that work "o" and Machine "0" are also on 0 - I start my cut, and after cutting - it doesn't return to "0" where it started - it moves this "0" about 20mm further.... Any ideas please???

2) When I power on my cnc, the motors go into LOCK - can't move the gantry by hand at all, I believe that when I connect to COM3 - that the motors are suppose to release - making the Gantry free to be moved by hand - NO SUCH LUCK - can't move the gantry as soon as it gets power - HOWEVER - gantry moves perfectly with software controller... is this right? - as previously, I was able to move the gantry by hand during any state of power and connected to COM 3 or not... 

thanks for the help! I do appreciate it!!
Andre

vMeph, 2017-05-18:
The setting $1 is where you set the time motors is enable if at 255 they always on, move motors with hand is not a good practice they generate energy, not good for the circuit i would say,you should jog machine to where you want to start job and mark a work position, read about work positions G54,55,56 etc..

Booysena, 2017-05-18:
Super - thanks vMeph!! something to work with is better than no idea at all :-) I'll be sure to check this out as suggested!! thanks a mill!!! :-)

cjbaar, 2017-05-18:
On issue #1... when you say it it "moves about 20mm further," does UGS show coordinates other than 0,0, or is the spindle now in the wrong place but UGS still thinks it is 0,0? Is it off on more than one axis?

Also, are you using the "Return to zero" command, or are there other commands at the end of your g-code file that are executing the return?

Booysena, 2017-05-19:
CJBaar: On issue 1 - before I give power to any motor / Arduino, I set the "0" bottom left, then I power up - and machine "0" reads as -20, work area "0" = "0" so my easiest solution was to "home" the spindle - and then move the work area to "0" so I kind'a bypassed the problem by moving my cutting area to "0"

ones the cut it done, it would return to "0" (which ends up being on the right spot now seeing as I've moved the cutting surface to "0"  

my thought is that everything is actually working correctly - compared to before (where it may not have been working correctly) before - I could set "0" on any point of the cutting surface, when when it was done cutting - it would return to THAT point, rather than the Machine "0" ..... I think I can live with things the way they are (as you can tell - I'm still learning, in South Africa there is very few people that do hobby CNC on components that I have on my machine - so it's very difficult to get help / ask questions.

guys, I appreciate your feedback and willingness to help! it's good to find questions as it allows one to think of into another direction that one has not thought of as yet - thus - Thanks You!! :-)

vMeph, 2017-05-19:
The -20 must come from $27 homing pull off settings?

winder, 2017-05-19:
The current work coordinate is modified with the "Reset Zero" button and it
is persisted in GRBL.

The `G10 L20` command is used:
http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G10-L20

Work coordinate overview:
https://www.shapeoko.com/wiki/index.php/G-Code#Using_the_Work_Coordinate_Systems

cjbaar, 2017-05-22:
In line with the other comments here, I always use a zero for my project, that is not "machine zero." This is what the work coordinates are designed to do. I agree that the -20 difference may be from the homing offset (I have mine set to only 3mm). Here is how I typically run my jobs.
- connect UGS
- enable the motor lock
- run homing sequence
- use jog or G91 to move the spindle to where I want the X/Y "zero" of my project to be
- click "Reset Zero"
- use a probe to finalize Z position zero
When running a gcode file, it will execute relative to the current work coordinates, so I always make sure the larger numbers on each axis under the Controller State panel are set properly to 0.00. The smaller numbers will still read absolute coordinates for your machine, but will mostly only come into play if you have soft limits set. If you have accurate homing, you can also use the absolute numbers to set a consistent and repeatable project home. Hope that makes sense.

Booysena, 2017-05-23:
Hi CJBaar, thanks for the info - yes, does make sense.... 
I've manage to resolve the issue of not returning to "0" - don't ask me how - I think all my fiddling did something :-) but something else came up - as it always does. When I add power to the motors, they don't lock, and from what I read in your comments, it appears that "Motor Lock" can either be enabled or disabled (2nd point you made) I'll focus on this a little and try to re-enable the motor locks. for the rest, I'm quite happy with how things are working and will hopefully not mess anything else up by trying the motor lock enable :-) 

thanks for all the guidance and feedback - much appreciated guys!!

cjbaar, 2017-05-23:
It depends on your controller. I happen to be using an arduino-based controller, and pin D8 is labeled as "stepper enable." On my setup, this pin is connected to a large toggle switch. I generally try to move the carriage by hand so it is close to home (just so homing doesn't take forever), and then enable the switch which "locks" the motors. It also functions as a kill switch. If something goes wrong with a job run for any reason, I can throw this switch and the motors will unlock and the controller stops responding to commands.

closed 2017-08-28 by winder