winder/Universal-G-Code-Sender #663: Feture Request

hexmaster111, 2017-05-24:
Would it be possible to add configurable keyboard shortcuts for every item in ugcs, even for jumping from text boxes.

winder, 2017-05-25:
@hexmaster111 there are keyboard shortcuts for almost everything in the UGS platform build in the configuration menu. What do you mean by jumping from text boxes, is tab not working for that?

hexmaster111, 2017-05-25:
I am thinking separate shortcuts to jump to spesfic boxes/the console. I am
thinking a button only interface/ no mouse. Thanks for the speedy response.

winder, 2017-05-25:
Unfortunately, I doubt I'll ever add this. You can get close - there are shortcuts you can set to open a specific window which I believe sets their focus.

closed 2017-05-25 by winder