winder/Universal-G-Code-Sender #665: Netbeans 8.1 starts but no UGS inside

MitchBradley, 2017-06-03:
UGS Platform nightly build as of 6/2/2017.  Latest JRE 8 as of same day.  Win 7 Pro 64 bit.

Running ugsplatform64.exe brings up the NetBeans Platform 8.1 window but there is no UGS application inside - it is just a big gray window with a generic NetBeans menu bar.

[image]

I tried changing the heap size in ugsplatform.conf; same results with 128m, 256m, 512m.

I turned on console logging and ran from a Command window, giving:

```
C:\Users\wmb\Downloads\ugsplatform\ugsplatform\bin>ugsplatform64.exe

C:\Users\wmb\Downloads\ugsplatform\ugsplatform\bin>

The launcher has determined that the parent process has a console and will reuse it for its own console output.
Closing the console will result in termination of the running program.
Use '--console suppress' to suppress console output.
Use '--console new' to create a separate console window.
-------------------------------------------------------------------------------
>Log Session: Friday, June 2, 2017 2:47:49 PM HST
>System Info:
  Product Version         = NetBeans Platform 8.1 (Build 201510222201)
  Operating System        = Windows 7 version 6.1 running on amd64
  Java; VM; Vendor        = 1.8.0_131; Java HotSpot(TM) 64-Bit Server VM 25.131-b11; Oracle Corporation
  Runtime                 = Java(TM) SE Runtime Environment 1.8.0_131-b11
  Java Home               = C:\Program Files\Java\jdk1.8.0_131\jre
  System Locale; Encoding = en_US (serverlaunch); Cp1252
  Home Directory          = C:\Users\wmb
  Current Directory       = C:\Users\wmb\Downloads\ugsplatform\ugsplatform
  User Directory          = C:\Users\wmb\AppData\Roaming\.ugsplatform\dev
  Cache Directory         = C:\Users\wmb\AppData\Roaming\.ugsplatform\dev\var\cache
  Installation            = C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform
                            C:\Users\wmb\Downloads\ugsplatform\ugsplatform\ugsplatform
                            C:\Users\wmb\Downloads\ugsplatform\ugsplatform\java
                            C:\Users\wmb\Downloads\ugsplatform\ugsplatform\extra
                            C:\Users\wmb\Downloads\ugsplatform\ugsplatform\ide
                            C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform
  Boot & Ext. Classpath   = C:\Program Files\Java\jdk1.8.0_131\jre\lib\resources.jar;C:\Program Files\Java\jdk1.8.0_131\
jre\lib\rt.jar;C:\Program Files\Java\jdk1.8.0_131\jre\lib\sunrsasign.jar;C:\Program Files\Java\jdk1.8.0_131\jre\lib\jsse
.jar;C:\Program Files\Java\jdk1.8.0_131\jre\lib\jce.jar;C:\Program Files\Java\jdk1.8.0_131\jre\lib\charsets.jar;C:\Progr
am Files\Java\jdk1.8.0_131\jre\lib\jfr.jar;C:\Program Files\Java\jdk1.8.0_131\jre\classes;C:\Program Files\Java\jdk1.8.0
_131\jre\lib\ext\access-bridge-64.jar;C:\Program Files\Java\jdk1.8.0_131\jre\lib\ext\cldrdata.jar;C:\Program Files\Java\
jdk1.8.0_131\jre\lib\ext\dnsns.jar;C:\Program Files\Java\jdk1.8.0_131\jre\lib\ext\jaccess.jar;C:\Program Files\Java\jdk1
.8.0_131\jre\lib\ext\jfxrt.jar;C:\Program Files\Java\jdk1.8.0_131\jre\lib\ext\localedata.jar;C:\Program Files\Java\jdk1.
8.0_131\jre\lib\ext\nashorn.jar;C:\Program Files\Java\jdk1.8.0_131\jre\lib\ext\sunec.jar;C:\Program Files\Java\jdk1.8.0_
131\jre\lib\ext\sunjce_provider.jar;C:\Program Files\Java\jdk1.8.0_131\jre\lib\ext\sunmscapi.jar;C:\Program Files\Java\j
dk1.8.0_131\jre\lib\ext\sunpkcs11.jar;C:\Program Files\Java\jdk1.8.0_131\jre\lib\ext\zipfs.jar
  Application Classpath   = C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\boot.jar;C:\Users\wmb\Downloads\
ugsplatform\ugsplatform\platform\lib\org-openide-modules.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib
\org-openide-util-lookup.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\org-openide-util-ui.jar;C:\User
s\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\org-openide-util.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform
\platform\lib\locale\boot_ja.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\locale\boot_pt_BR.jar;C:\Us
ers\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\locale\boot_ru.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform
\platform\lib\locale\boot_zh_CN.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\locale\org-openide-modul
es_ja.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\locale\org-openide-modules_pt_BR.jar;C:\Users\wmb\
Downloads\ugsplatform\ugsplatform\platform\lib\locale\org-openide-modules_ru.jar;C:\Users\wmb\Downloads\ugsplatform\ugsp
latform\platform\lib\locale\org-openide-modules_zh_CN.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\lo
cale\org-openide-util-lookup_ja.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\locale\org-openide-util-
lookup_pt_BR.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\locale\org-openide-util-lookup_ru.jar;C:\Us
ers\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\locale\org-openide-util-lookup_zh_CN.jar;C:\Users\wmb\Downloads\u
gsplatform\ugsplatform\platform\lib\locale\org-openide-util-ui_ja.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\pla
tform\lib\locale\org-openide-util-ui_pt_BR.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\locale\org-op
enide-util-ui_ru.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\locale\org-openide-util-ui_zh_CN.jar;C:
\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\locale\org-openide-util_ja.jar;C:\Users\wmb\Downloads\ugsplatf
orm\ugsplatform\platform\lib\locale\org-openide-util_pt_BR.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\l
ib\locale\org-openide-util_ru.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\lib\locale\org-openide-util_zh
_CN.jar;C:\Program Files\Java\jdk1.8.0_131\lib\dt.jar;C:\Program Files\Java\jdk1.8.0_131\lib\tools.jar
  Startup Classpath       = C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\core\asm-all-5.0.1.jar;C:\Users\wmb\
Downloads\ugsplatform\ugsplatform\platform\core\core-base.jar;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform\co
[... 115 lines cut ...]
        -Xmx768m
        -XX:MaxPermSize=384m
        -Dnetbeans.user.dir=C:\Users\wmb\Downloads\ugsplatform\ugsplatform\bin
        -Djdk.home=C:\Program Files\Java\jdk1.8.0_131
        -splash:C:\Users\wmb\AppData\Roaming\.ugsplatform\dev\var\cache\splash.png
        -Dnetbeans.home=C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform
        -Dnetbeans.user=C:\Users\wmb\AppData\Roaming\.ugsplatform\dev
        -Dnetbeans.default_userdir_root=
        -XX:+HeapDumpOnOutOfMemoryError
        -XX:HeapDumpPath=C:\Users\wmb\AppData\Roaming\.ugsplatform\dev\var\log\heapdump.hprof
        -Dsun.awt.keepWorkingSetOnMinimize=true
        -Dnetbeans.dirs=C:\Users\wmb\Downloads\ugsplatform\ugsplatform\platform;C:\Users\wmb\Downloads\ugsplatform\ugspl
atform\ugsplatform;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\java;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\ex
tra;C:\Users\wmb\Downloads\ugsplatform\ugsplatform\ide
        exit
Compiler: HotSpot 64-Bit Tiered Compilers
Heap memory usage: initial 512.0MB maximum 683.0MB
Non heap memory usage: initial 2.4MB maximum -1b
Garbage collector: PS Scavenge (Collections=1 Total time spent=0s)
Garbage collector: PS MarkSweep (Collections=1 Total time spent=0s)
Classes: loaded=4390 total loaded=4390 unloaded 0
INFO [org.netbeans.core.ui.warmup.DiagnosticTask]: Total memory 8,471,691,264
INFO [null]: Total physical memory 8,471,691,264

```

winder, 2017-06-03:
@MitchBradley very strange, I don't see any errors in the log. If you open up the program again and go to the `about` page you'll see a `User directory` and `Cache directory`. Try deleting those directories and reloading to see if it helps.

MitchBradley, 2017-06-03:
Thanks, deleting the User directory (dev), which included the Cache
directory as a subdirectory (dev/var/cache), solved the problem.

I have an idea about how the problem may have come about.  The first
time I tried to run the program, I had JRE 8 installed, but there was
also an older JDK 7 installation on my computer.  I think that the
program may have chosen the old JDK instead of the newer JRE, and then
failed, possibly leaving bad data in the User/Cache directories.  The
reason I suspect this is because I saw a reference to jdk 1.7 in an
earlier copy of the log.

closed 2017-06-13 by winder