winder/Universal-G-Code-Sender #668: Macros pane does not function

cjbaar, 2017-06-10:
I apologize in advance for re-posting this, but I couldn't figure out how to re-open an older post, and this issue does not appear to have changed. I am using UGS platform. I am downloading using the link shown here, under the "Nightly Builds" section, as it seems to be the most relevant I can find. I am unable to find any version information in the application (other than the version of Java/NetBeans) so I don't know how to tell if it is current or not.
[image]

This problem has persisted since I started using UGS. I am creating macros using the preferences window.
[image]

This creates a button in the Macros pane.
[image]

However, clicking on this button does nothing. No commands are sent, as shown in the console output here.
[image]

I would appreciate any advice on what I may be missing. I have gone through this several times, and re-download UGS each time but no changes appear to have been made. Thank you.

poisa, 2017-06-15:
Just came here to report the same.

winder, 2017-06-20:
@cjbaar @poisa first of all, sorry for the delay. I reviewed your comments in #636 and haven't been able to reproduce the issue yet.

What version of GRBL are you connecting to?
What OS are you using?
If you launch UGS from the command line are any errors reported?
When you click the macro button do you see a red symbol in the bottom right corner of the platform GUI? If yes please click it and copy the error message here.

One thing you can try is to delete the cache directory, you can find its location on the `About` page:
http://winder.github.io/ugs_website/img/platform/about_popup.png

winder, 2017-06-20:
Some other questions occurred to me:

do the macro button in the settings panel work?
do the macro menu items work?

vMeph, 2017-06-20:
try delete the temps data hide?
\AppData\Roaming.ugsplatform

poisa, 2017-06-20:
> What version of GRBL are you connecting to?

1.1

> What OS are you using?

macOS Sierra 10.12.5 (16F73)

> If you launch UGS from the command line are any errors reported?

No. Just this:
```
Loading LocalizingService...
Initializing...
Loading JogService...
Loading ActionService...
Loading MacroService...
Loading StatusLineService...
Services loaded!
Initializing firmware... ...
/resources/firmware_config/
/resources/firmware_config/tinyg.json
/resources/firmware_config/smoothieboard.json
/resources/firmware_config/test_1.json
/resources/firmware_config/grbl.json
/resources/firmware_config/test_2.json
Setting changed
Is enabled: false
```

> When you click the macro button do you see a red symbol in the bottom right corner of the platform GUI? If yes please click it and copy the error message here.

No.

> One thing you can try is to delete the cache directory, you can find its location on the About page:
http://winder.github.io/ugs_website/img/platform/about_popup.png

I don't have that dir. I mean, I have all the tree except for the cache part. Could this probably be because I just re-installed the nightly build and haven't connected it to the CNC yet?

> Do the macro button in the settings panel work?

Didn't try this. I will as soon as I'm near the CNC this afternoon.

> Do the macro menu items work?

They change their label and are enabled but nothing seems to happen when I click them (the logs don't show anything and there are no error messages anywhere).

poisa, 2017-06-21:
>Do the macro button in the settings panel work?

Just tried this and they don't work either.

I re-tried the same steps with an older version of UGS and the latest nightly build both with same results.

The GRBL version is actually `1.1f` (not `1.1` as I stated before).

poisa, 2017-06-21:
I installed the latest version on a different machine which never had UGS before and finally got something that might be useful to debug this. After touching one of the macro buttons on the main panel (not the preferences panel) I got a red "minus" sign at the bottom with the following:

java.lang.NoSuchMethodError: com.willwinder.universalgcodesender.model.Position.getX()D
	at com.willwinder.universalgcodesender.model.GUIBackend.updateSystemState(GUIBackend.java:365)
	at com.willwinder.universalgcodesender.MacroHelper.substituteValues(MacroHelper.java:86)
	at com.willwinder.universalgcodesender.MacroHelper.executeCustomGcode(MacroHelper.java:46)
	at com.willwinder.universalgcodesender.uielements.macros.MacroActionPanel.customGcodeButtonActionPerformed(MacroActionPanel.java:170)
	at com.willwinder.universalgcodesender.uielements.macros.MacroActionPanel.lambda$doLayout$0(MacroActionPanel.java:100)
	at com.willwinder.universalgcodesender.uielements.macros.MacroActionPanel$$Lambda$63/943909185.actionPerformed(Unknown Source)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6525)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3321)
	at java.awt.Component.processEvent(Component.java:6290)
	at java.awt.Container.processEvent(Container.java:2234)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2739)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:746)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:719)
	at java.awt.EventQueue$4.run(EventQueue.java:717)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:716)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:159)
[catch] at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

cjbaar, 2017-06-24:
My grbl is 0.9j, desktop is OS X 10.10.5. When I click on the macro, I see the red icon, and it provides the following. At first glance, it appears the same as @poisa 

java.lang.NoSuchMethodError: com.willwinder.universalgcodesender.model.Position.getX()D
	at com.willwinder.universalgcodesender.model.GUIBackend.updateSystemState(GUIBackend.java:366)
	at com.willwinder.universalgcodesender.MacroHelper.substituteValues(MacroHelper.java:86)
	at com.willwinder.universalgcodesender.MacroHelper.executeCustomGcode(MacroHelper.java:46)
	at com.willwinder.universalgcodesender.uielements.macros.MacroActionPanel.customGcodeButtonActionPerformed(MacroActionPanel.java:170)
	at com.willwinder.universalgcodesender.uielements.macros.MacroActionPanel.lambda$doLayout$0(MacroActionPanel.java:100)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
	at java.awt.Component.processMouseEvent(Component.java:6535)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6300)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4891)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:159)
[catch] at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

cjbaar, 2017-06-24:
I have deleted the cache directory, but no change. The macros button in the settings panel does not work, and I have no items showing the macros menu.

vMeph, 2017-06-24:
did you delete the temp files hiden ??
 C:\AppData\Roaming\.ugsplatform

poisa, 2017-06-24:
I'm not no Windows, so no.

>

sdarocha, 2017-10-20:
@cjbaar and @poisa did you solved the problem? Same problem with me. Thank you.

poisa, 2017-10-20:
No, I just learned to live with it. I don’t do this professionally so it’s
ok.

winder, 2017-10-20:
I missed the stacktrace from @cjbaar back in June, I just added a fix for it.

sdarocha, 2017-10-20:
@poisa I was trying the "Somewhat stable" UGS Platform. I thought it was the best one.
Now I try the UGS Platform "Normal" and it works ok.
Thank you @winder

poisa, 2017-10-20:
@winder That's great to hear. Will give this a try soon.

breiler, 2018-06-18:
Closing this issue as some of the users now reports that this is fine.

closed 2018-06-18 by breiler