winder/Universal-G-Code-Sender #670: Platform wont launch in Win 7

dblaron30i, 2017-06-13:
So USGplatform doesnt seem to work on my Win7 machine. I tried it on my Mac works fine. I tried it on Win10 64bit and works fine. I did do a little research and found that Netbean will run on my Win7 machine. Java is the latest and same version as the Win 10 copy. I am running Win7 64bit. The App just does not launch at all. I am using from the Bin folder. On other thing is that the USG Classic works perfect on the Win 7 Machine. I just would love to use all the new features from  the Platform version. Any ideas what it could be?

winder, 2017-06-13:
@dblaron30i if you run the startup script from the command line do you see any errors in the console? Please copy the output and put it here, let me know if you need help running from the command line.

dblaron30i, 2017-06-13:
where is the setup script located in the folders of USGplatform. I will try this next.

winder, 2017-06-14:
@dblaron30i oops, this is only a script for linux/mac. In `ugsplatform/bin/` run `ugsplatform.exe` or `ugsplatform64.exe` from the command prompt, it should have more information about what's going on here.

dblaron30i, 2017-06-14:
Ok so here is what I found. 

The launcher has determined that the parent process has a console and will reuse it for its own console output.
Closing the console will result in termination of the running program. 
Use '--console suppress' to suppress console output.
Use '--console new' to create a separate console window.
Error occurred during initialization of VM
Initial heap size set to a larger value than the maximum heap size.

dblaron30i, 2017-06-14:
Hope this thread will help in finding a resolution. I looked on your page and no minimum memory is listed that I could find. Netbean says it will run on as little as 512mb on win 7 so maybe that's why the issue went un-noticed by me. 

https://stackoverflow.com/questions/9219206/what-does-mean-maximum-heap-size-unlimited-in-a-j2me-device

winder, 2017-06-14:
@dblaron30i yes that helps. This is the same problem as issue #615, I thought I was waiting for some feedback on that one but it looks to be completely resolved and I need to update one of the default settings. 

In the **etc** directory, open **ugsplatform.conf** and change the **default_options** value:
```
default_options="--branding serverlaunch -J-Xms128m"
```

dblaron30i, 2017-06-14:
Cool thanks I really appreciate it. I really like the versatility of the new version. I am trying to run dual monitors and one as a touch screen for jogging and some other functions. So this helps.

dblaron30i, 2017-06-14:
Ok so in your code you had -Xms256m  I Changed it to -Xmx1g and the program loaded perfectly in Win7. 
Thanks.

winder, 2017-06-15:
@dblaron30i thanks for the update!

closed 2017-06-15 by winder