winder/Universal-G-Code-Sender #671: Switch from JOGL to JavaFX 3D

winder, 2017-06-14:
For better compatibility, easier maintenance, convenient high-level APIs, and smaller binaries, UGS should use the high-level JavaFX 3D libraries that are built into Java8+ instead of the low-level JOGL libraries.

vMeph, 2017-06-15:
@winder  what this means?
this will add better 3d capabilities, like for example use visualizer with better 3d view graphics?

winder, 2017-06-15:
@vMeph JOGL is a low-level library while JavaFX 3d is a high-level library. You can sort of think of it like calligraphy vs. typing into a computer with a nice font. Calligraphy can give you incredible if you're good at it but it is a lot of effort and if you aren't good it shows, while typing into a text editor is going to give you great results even if you're not an expert.

There are small things in the low-level approach I have which are difficult and take a lot of time to figure out, for example in the new autoleveler surface preview the surface isn't transparent for some reason.

Also I jump through a lot of hoops to get things to work on different platforms by including lots of libraries with JOGL, all that is built into the java runtime for Java FX.

closed 2017-11-30 by winder