winder/Universal-G-Code-Sender #672: Universal GcodeSender Pendant

vMeph, 2017-06-15:
@winder  was looking into the new icon for Universal GcodeSender Pendant
wow this a great addition to UGS was testing control machine from my smartphone looking good
one thing i see that isn't following is the controls feedrate, one the program you set the feedrate if you move machine it will do the feedrate as expected , but on the smartphone it will not follow the same feedrate that is set on the UGS program.
also it would be great to add a press continuity button for when you moving the machine, for example if i would keep the finger on X button it should keep moving until you release, the way it is now  if you keep finger on it will only move one time and stops.

would it possible to add like also a camera view?

looking good great job!!!

winder, 2017-06-15:
@vMeph the funny/sad thing about this... the Pendant has been in UGS for the past couple years and I never got around to adding the button. I don't know much about coding web pages so any enhancements here need to come from the community.

@acoll pendant feature request, what do you think?

vMeph, 2017-06-16:
@winder wish i could help on this but my coding skills not that good to help out, but for sure this is a great thing to have on the UGS in web interface would be cool or even if was a android application could rox,
this would give a better control for the machine 
you could use has a joystick if was added continuity to the buttons already implemented
if would have a command sender could be cool
and a camera wold make wonders, specially when you doing a job and you need to step out fast, then you can view what is happening while you leave your machine unattended shortly.

would be great if community would give a help to make it happen

vMeph, 2017-06-25:
@winder  whats name of file that is implemented the code for the GcodeSender?

vMeph, 2017-06-27:
@winder by going to pendant page http:  i press control+U you can see the code that is used to do the controls 
this is the code
[pendant.txt](https://github.com/winder/Universal-G-Code-Sender/files/1106951/pendant.txt)
in what file inside UGS folder you find where this is code is written?
maybe you could take a look see if anything could be done to improve some things?
the options off having this option have  alot potencial for UGS sense you can control machine and keep  a eye on machine actions on a android cellphone etc..

acoll, 2017-06-27:
@vMeph the client side code for the client can be found here: https://github.com/winder/Universal-G-Code-Sender/tree/master/ugs-core/src/resources/pendantUI

I've played around with it a little but and the settings for the pendant dont seem to match what the UGS jogger does. I'm going to look at it in the next couple weeks as i get free time to play with the CNC.

vMeph, 2017-06-27:
@acoll cool
would be nice if we could add the jog to function while keeping button press on, right now only does 1 step and stops
if could be done a tab to add a camera from com could be also a cool feature for we be able to watch what machine is doing from the android

vMeph, 2017-06-28:
@acoll maybe while button is press, would need to be on some  milli seconds then go off verify state of button and go on again some mS while then button keeps pressed would keep doing same? 

unfortunately my coding skills not that good specially on java script :(

vMeph, 2017-06-29:
it seems that jog buttons also is ignoring FeedRate that you set on the main program

vMeph, 2017-06-29:
i was able to test with a java webcam tutorial to acess a phone web cam,
basikly now a days we all have a old smartphone laying arround so if you install a App like webcam ip on smart phone then you could acess a web cam from Gcode sender pendant, i was able to test it with netbeans and worked, but unfortunatly my coding is very bad and i dont know how to add another button to pendant with a sec tion for Cam Acess and with a layout that would have a place to insert the desire ip of the smartphone to run the app

the tutorial i used was on this video
https://www.youtube.com/watch?v=3_GNE0t_VzY

the libary code of java is 
https://github.com/Junaid67/IP-Camera-Java-CV-Libraries
and the code link
https://github.com/Junaid67/Access-Ip-Camera-using-java-code

winder, 2017-10-13:
Closing this for now since the current jogger probably needs to be completely replaced before anything specific can be improved about it.

closed 2017-10-13 by winder