winder/Universal-G-Code-Sender #673: stepper motors not working through UGS?

joew2840, 2017-06-15:
I have just finished setting up a 2 axis cnc/printer for a school project of mine. I am using an arduino and a generic v2 cnc shield. I am not really sure what drivers it uses and may upgrade them later if needed. These run 2 nema 17 stepper motors. However when beginning to test out the electronics and make sure everything was working, I ran into a problem.

 When sending a single gcode command through the arduino, the stepper motors would each turn the set amount they had to turn. However when I used UGS to send a gcode script (just a simple circle to start off with) the stepper motors would just make a horrible screeching noise but would not move. 

To begin with I thought that maybe UGS is sending the gcode commands too quickly, and the steppers cannot keep up. Is there anyway to change the rate which commands are sent through UGS?

After some research however I also discovered that I may not be supplying enough voltage to the arduino.  

If anyone else has some ideas as to what is the cause, I am all ears. Thanks

vMeph, 2017-06-15:
what version grbl you using?
i dont know  the generic v2 cnc shield but the V3 for exemple was made for versions of GRBL 0.8 below so if you using like version 0.0 above you may need to swap 2 pins sense there was changes on the new versions check GRBL wiki to see those changes.
about motors dont move and make noise, im assuming that you have drivers A4988 or DRV8825, did you adjust the trimpot on those drivers to the proper current that you need to supply to the motors?
sounds that you might not have the drivers adjust to the proper current

joew2840, 2017-06-15:
@vMeph  Thank you so much! I am not able to test the pot settings yet, but as soon as I get the chance I definitely will. I believe that I am using grbl 0.8 and a v3 shield with a4988 drivers. Also what do you mean by swapping the 2 pins sense? Thanks

winder, 2017-06-16:
Joe, I haven't tested 0.8 in a long time. You might also want to try older
versions of UGS.

That said, screeching motors usually means you have it wired wrong. If you
don't have the motor specs you can figure out which wires are pairs with
trial and error, or short them together until you find which pairs of wires
make the stepper harder to turn.

vMeph, 2017-06-16:
@joew2840 what i mean by pins swap is that the CNC shield v3 was done for the GRBL version 0.8 and below and the versions GRBL 0.9 above there was changes made to the arduino pins, pin 11 and 12 was changed, so if you plan on later have the latest version of GRBL  1.1 with the new features you will need to change those pins, that can be easy done by  soldering on bottom of arduino some cables on pin 11 and 12 and then bend on the CNC shield the pin 11 and 12 and  connect the arduino pin 11 to the 12 of CNC shield and the arduino pin 12 to the 11 of the CNC shield, you may want to take a look on the changes that was made on this page
https://github.com/grbl/grbl/wiki/Connecting-Grbl

about your problem you can do like Winder suggested and with a multimeter see witch wires have continuity to see whats the coils of the motor, 2B 2A is one coil and 1A 1B is the other

[image]

closed 2017-06-18 by winder