winder/Universal-G-Code-Sender #674: toolbar missing buttons..........help

AEATONTCG, 2017-06-17:
the connection, startpausestop, pendant, gcodebrowser toolbars or showing up as <connection> ect. therefore I cant connect my cnc machine. I'm not sure if this is a bug or if its on my end please help

winder, 2017-06-17:
1) Make sure you have Java8 installed.
2) If you right click the toolbar there is a `Reset Toolbars` button that might fix it.
3) As a last resort clearing the cache will fix it, in the "About" page is a link to the cache directory on your machine.

AEATONTCG, 2017-06-17:
I have java 8 and tried resetting and it didnt work, I went and deleted all the files that I could in the cache and that didnt work

AEATONTCG, 2017-06-17:
I have java 8 and tried to reset and deleted the cache folder nothing worked

AEATONTCG, 2017-06-17:
@winder  I have java 8 and reset didn't work, and clearing the cache folder didn't work. I've uninstalled and reinstalled 3 times and its the same result every time, please help, I'm green to cnc and everyone is telling me to use this program.

vMeph, 2017-06-17:
@AEATONTCG  did you try delete the temps data hide?
\AppData\Roaming\.ugsplatform

AEATONTCG, 2017-06-18:
@vMeph. Thank you, Thank you! that did the trick

vMeph, 2017-06-18:
@AEATONTCG  you welcome mate
You may close the issue

closed 2017-06-18 by AEATONTCG