winder/Universal-G-Code-Sender #675: Already Streaming Problem

trbLeeciN, 2017-06-19:
Hello, I ended with an other problem. When I try to send files many times (one by one). I sometimes receive the problem that says "Already streaming." when trying to send file again.
I edited followings;

    @Override
    public void commandSkipped(GcodeCommand command) {
    	System.out.println("skipped : "+ this.backend.getNumRemainingRows() + " // "+ this.backend.getNumRows());
    }

    @Override
    public void commandComplete(final GcodeCommand command) {
    	System.out.println("complete : "+ this.backend.getNumRemainingRows() + " // "+ this.backend.getNumRows());
    }

    @Override
    public void fileStreamComplete(String filename, boolean success) {
        System.out.println(""+success);
    }
    
to check whether remaninrows become 0 or not. Even remainingrows turn into 0, filestreamcomplete not called so I can not continue sending files.
Can you please suggest any solutions for this? Thank you very much for suggestions and your previous solutions..

winder, 2017-06-19:
http://winder.github.io/ugs_website/dev/backend_development/#communicator

Are you implementing a new controller, or using the existing GRBL controller? If you're using GRBL, what version of GRBL?

trbLeeciN, 2017-06-19:
Well I am using existing GRBL controller I guess. and I am using grbl 1.1

winder, 2017-06-19:
Is your code on github? It might be easier if I review how you're using it.

You can review `MainWindow.java`, or grep for `backend` in the ugs-platform directory for all the ways it is used in the UGS platform to accomplish specific tasks.

trbLeeciN, 2017-06-19:
Well I am very new to coding things. but my code is not long, so i would be very happy if you can check it from here. 
import com.willwinder.universalgcodesender.utils.Settings;
import com.willwinder.universalgcodesender.utils.SettingsFactory;
import com.willwinder.universalgcodesender.model.BackendAPI;
import com.willwinder.universalgcodesender.model.UGSEvent;
import com.willwinder.universalgcodesender.model.UGSEvent.ControlState;
import com.willwinder.universalgcodesender.listeners.ControllerListener;
import com.willwinder.universalgcodesender.listeners.ControllerStatus;
import com.willwinder.universalgcodesender.listeners.UGSEventListener;
import com.willwinder.universalgcodesender.types.GcodeCommand;
import java.io.File;
import java.util.logging.LogManager;
import java.util.logging.Logger;

public class UGS implements ControllerListener, UGSEventListener {
    
	public Settings settings;
    BackendAPI backend;
    String gcodeFile = null;
    String processedGcodeFile = null; 
    String srp="ready";
    @SuppressWarnings("unchecked")

    public static void main(String args[]) {
        LogManager.getLogManager().reset();
        Logger globalLogger = Logger.getLogger(java.util.logging.Logger.GLOBAL_LOGGER_NAME);
        globalLogger.setLevel(java.util.logging.Level.OFF);
        new UGS();
    }
 
    /** Creates new form MainWindow */
    public UGS() {
    	LogManager.getLogManager().reset();
        Logger globalLogger = Logger.getLogger(java.util.logging.Logger.GLOBAL_LOGGER_NAME);
        globalLogger.setLevel(java.util.logging.Level.OFF);
    	GUIBackend backend = new GUIBackend();
        this.backend = backend;
        this.settings = SettingsFactory.loadSettings();
        try {
            backend.applySettings(settings);
        } catch (Exception e) {}
        
        backend.addControllerListener(this);
        backend.addUGSEventListener(this);
        
        Runtime.getRuntime().addShutdownHook(new Thread() {
            @Override
            public void run() {
                settings.setPort("/dev/ttyACM0");
                settings.setPortRate("115200");
                settings.setFirmwareVersion("GRBL");
                SettingsFactory.saveSettings(settings);
            }
        });
        try {
            this.backend.connect("GRBL", "/dev/ttyACM0", 115200);
            Thread.sleep(8000);
        } catch (Exception e) {}
    }  
    
    public void sendCommand(String command) throws Exception {//GEN-FIRST:event_sendButtonActionPerformed
    	srp="ready";
    	this.backend.sendGcodeCommand(command);
    }//GEN-LAST:event_sendButtonActionPerformed
    
    
    public void sendGcodeFile(File gcodeFile) {//GEN-FIRST:event_sendButtonActionPerformed
        try {
        	backend.setGcodeFile(gcodeFile);
            this.backend.send();
        	srp="process";
        } catch (Exception e) {}
        
    }//GEN-LAST:event_sendButtonActionPerformed

    public void sendPauseResume(String state) {
    	try {
    		srp=state;
            this.backend.pauseResume();
        } catch (Exception e) {
        	System.out.println("Pause/Resume Failed");
        }
    }
    public String returnsrp(){
    	return srp;
    }
    
    @Override
    public void fileStreamComplete(String filename, boolean success) {
        System.out.println(""+success);
    }
    
    @Override
    public void controlStateChange(ControlState state) {}

    @Override
    public void commandSkipped(GcodeCommand command) {
    	System.out.println("skipped : "+ this.backend.getNumRemainingRows() + " // "+ this.backend.getNumRows());
    }
     
    @Override
    public void commandSent(final GcodeCommand command) {}
    
    @Override
    public void commandComment(String comment) {
    }
    
    @Override
    public void commandComplete(final GcodeCommand command) {
    	System.out.println("complete : "+ this.backend.getNumRemainingRows() + " // "+ this.backend.getNumRows());
    }
    @Override
    public void messageForConsole(final MessageType type, final String msg) {}
    
    @Override
    public void statusStringListener(ControllerStatus status) {}
    
    @Override
    public void postProcessData(int numRows) {}
    
    @Override
    public void UGSEvent(UGSEvent evt) {}
    
    
}

winder, 2017-06-19:
This is a little tricky but I'll do my best to describe the situation.

When GRBL has processed a message from it's receive-buffer it returns `ok`. This doesn't mean that the command has finished executing but does mean that more commands may be sent by the GUI. In UGS when an `ok` is received the `commandComplete` callback is triggered and on the last command, you'll see `getNumRemainingRows` returns zero. So even though UGS says there is nothing remaining, there may still be active commands finishing up.

So what to do about it. The easiest way to update your programs state is with the `UGSEvent` callback. That is the "newest" and thus the best thought out of the callbacks. In my modules I'll usually have something like this:
```
    @Override
    public void UGSEvent(UGSEvent evt) {
        if(evt.isSettingChangeEvent()) {
            updateSettings();
        }
    }
```

Inside `updateSettings()` I'll use these helper methods to determine what to enable:
```
    // Controller status
    public boolean isConnected();
    public boolean isActive();
    public boolean isSendingFile();
    public boolean isIdle();
    public boolean isPaused();
    public boolean canPause();
    public boolean canCancel();
    public boolean canSend();
    public ControlState getControlState();
```

Specifically to your question about "when can I send another file", here is how I enable the "send" button (it does double-duty for send and resume):
```
    @Override
    public boolean isEnabled() {
        return backend.canSend() || backend.isPaused();
    }

    @Override
    public void actionPerformed(ActionEvent e) {
        try {
            if (backend.isPaused()) {
                backend.pauseResume();
            } else {
                backend.send();
            }
        } catch (Exception ex) {
            GUIHelpers.displayErrorDialog(ex.getLocalizedMessage());
        }
    }
```

trbLeeciN, 2017-06-20:
Thanks for your help. I am not a good enough coder to understand and implement your thoughts into my project. I decided to try using softreset in order to solve my problem. It is probably really noobish but it seems to be working.

 @Override
    public void commandComplete(final GcodeCommand command) {
    	System.out.println("complete : "+ this.backend.getNumRemainingRows() + " // "+ this.backend.getNumRows());
    	if(this.backend.getNumRemainingRows()==0){
    		try {
				System.out.println(this.backend.isSendingFile());
				//this.backend.
				backend.issueSoftReset();
			} catch (Exception e) {
			}
    	}
    }

winder, 2017-06-20:
@trbLeeciN you just need something like this and you'll only be able to click the send button at the correct time:

```
    @Override
    public void UGSEvent(UGSEvent evt) {
        if(evt.isSettingChangeEvent()) {
            <<your send button>>.setEnabled(backend.canSend());
        }
    }
```

closed 2017-06-21 by winder