winder/Universal-G-Code-Sender #679: Ugs visualizer and gcode errors out of nowhere

allcustomz, 2017-06-26:
my ugs and gcode files all worked perfectly i try using new software deskcnc to try to get a picture i couldn't get into gcode due to bad picture. Soo after transferring in fengrave and ugs it would pop up on visualizer then disappear?? Then it would say bad format all gcode? Nc correct too? Now all my gcode files even old one wont work on visualizer and ugs bad format as well. The old files worked a million times before now i cant do anything

winder, 2017-06-26:
Could you share one of the files which aren't working?

allcustomz, 2017-06-26:
Sorry heres the rest of my comment so some files work and some dont dissipers and errors in code but the files never had problems before till today? What is going on? Thank you for anybodys time .

allcustomz, 2017-06-26:
Yes as a picture? Also the files weren't curropt before eather but ilk send a picture

winder, 2017-06-26:
Please attach one of the gcode files which aren't working to the ticket

allcustomz, 2017-06-26:
Winder @winder ""Error was detected while sending gz.2 to error bad format missing the expected G-Code word value or numeral you format is not valid streaming has been paused"" this goes on for ALL the code in the file and the file was not corrupt before any ideas? I had this error before but my gcode was actually no good? Now its outta the blue

winder, 2017-06-26:
My guess is that there is an electronics issue interfering with the file streaming, this could explain why the issue started happening suddenly. If you are using an Arduino clone there are some known communication issues which could also cause this: https://github.com/gnea/grbl/wiki/Known-Issues#usb-to-serial-transmission-errors

Please attach one of the bad files to this ticket so I can verify that the file is good. You can drag/drop the file into the text box to attach it.

allcustomz, 2017-06-26:
had to copy ill send another to       
(##########################################)
G90
G20
G17 G90 G64 M3 S3000 M7 
G0 Z0.2500
G0 X4.2062 Y2.1204
G1 Z-0.0630 F235.0000
G1 X4.3230 Y2.0954
G1 X4.4110 Y2.0325
G1 X4.4377 Y2.0107
G1 X4.4524 Y1.9920
G1 X4.4632 Y1.9709
G1 X4.4699 Y1.9481
G1 X4.4722 Y1.9145
G1 X4.4722 Y1.6629
G1 X4.4700 Y1.6298
G1 X4.4637 Y1.6075
G1 X4.4533 Y1.5868
G1 X4.4393 Y1.5684
G1 X4.4135 Y1.5464
G1 X4.3704 Y1.5140
G1 X4.3360 Y1.4926
G1 X4.2973 Y1.4826
G1 X4.2369 Y1.4725
G1 X4.2188 Y1.4679
G1 X3.7341 Y1.2968
G1 X3.6935 Y1.2863
G1 X3.6231 Y1.2746
G1 X3.0237 Y1.0749
G1 X3.0212 Y1.0707
G1 X3.0231 Y1.0390
G1 X3.0231 Y0.5862
G1 X3.0331 Y0.5810
G1 X3.0641 Y0.5834
G1 X3.0811 Y0.5866
G1 X3.1551 Y0.6077
G1 X3.2049 Y0.6159
G1 X3.9093 Y0.6452
G1 X3.9473 Y0.6441
G1 X4.3652 Y0.6008
G1 X4.4102 Y0.5914
G1 X4.8590 Y0.4466
G1 X4.8892 Y0.4347
G1 X4.9080 Y0.4224
G1 X4.9242 Y0.4069
G1 X4.9374 Y0.3887
G1 X4.9472 Y0.3685
G1 X4.9532 Y0.3469
G1 X4.9552 Y0.3132
G1 X4.9552 Y0.2906
G1 X4.9533 Y0.2578
G1 X4.9479 Y0.2371
G1 X4.9389 Y0.2176
G1 X4.9267 Y0.1999
G1 X4.9117 Y0.1845
G1 X4.8943 Y0.1719
G1 X4.8899 Y0.1626
G1 X4.9073 Y0.1512
G1 X4.9225 Y0.1371
G1 X4.9352 Y0.1207
G1 X4.9449 Y0.1024
G1 X4.9516 Y0.0827
G1 X4.9548 Y0.0623
G1 X4.9537 Y0.0311
G1 X4.9405 Y-0.1142
G1 X4.8340 Y-0.2421
G1 X4.8305 Y-0.2518
G1 X4.8177 Y-0.2813
G1 X4.8052 Y-0.2992
[... 1948 lines cut ...]
G1 Z-0.0630
G1 X-3.3255 Y-1.5318
G1 X-3.3095 Y-1.5943
G1 X-3.5691 Y-1.5943
G0 Z0.2500
G0 X-2.3562 Y1.0903
G1 Z-0.0630
G1 X-0.7514 Y1.0903
G0 Z0.2500
G0 X-0.8888 Y1.0278
G1 Z-0.0630
G1 X-2.2577 Y1.0278
G1 X-2.1675 Y0.9654
G1 X-1.0322 Y0.9654
G0 Z0.2500
G0 X-1.1472 Y0.9030
G1 Z-0.0630
G1 X-1.9523 Y0.9030
G1 X-1.8552 Y0.8405
G1 X-1.2284 Y0.8405
G1 X-1.3221 Y0.7781
G1 X-1.5752 Y0.7781
G0 Z0.2500
M5
M2

vMeph, 2017-06-26:
or maybe? when some of the errors appears did you say yes to ignore those commands that use to work in  files and now doesnt .
on the  Tools/Options/UGS/ Controller Option is there now  list to ignore those commands and now is enable to be ignored on the Regular Expression Pattern Remover?
cause if was working before strange that now doesnt

allcustomz, 2017-06-26:
@vmeph yes i hit ignore commands in future but its 4000 or how ever many of the boxs pop up? Soo what do you think i should do erase the regular expression pattern remover and remove it from the enabled section?

vMeph, 2017-06-26:
yes remove them from there and give a test

allcustomz, 2017-06-26:
@vmeph wow it might have worked!!!!! thank you so much! if it keeps happing ill ask. but it seems to work good. I wonder why it did that? thank you.

vMeph, 2017-06-26:
no idea why happen, most like is what winder mention, some bad communication happen and then you say to ignored all to those commands

allcustomz, 2017-06-26:
@winder @vmeph  yea you guys are great thank you sorry about the long comment winder and vmeph but thanks again to both of you

allcustomz, 2017-06-26:
@winder @vmeph should I close this thread Now? This solved my issue

winder, 2017-06-26:
@allcustomz great, glad that resolved the issue. Thanks @vMeph for figuring it out!

This issue has happened a few times now, I think it would be useful to add a popup after loading a file that says something like `Skipped XXX lines of gcode with the pattern remover, verify sender settings if this is not expected.`

closed 2017-06-26 by winder

vMeph, 2017-06-26:
@winder that might not be a bad idea

rmisko11, 2017-06-26:
This would be helpful, the same thing has happened to me in the past and
took a while to resolve.

vMeph, 2017-06-26:
@winder  that popup warning saying something could then be add to the Tools/Options/UGS/ Controller Option, has a check box that you can enable or disable  popup verify sender option.

vMeph, 2017-06-26:
@allcustomz  that block on your file startup that you add  here of your Gcode file
G17 G90 G64 M3 S3000 M7
G64 i dont think thats a command suported by GRBL
at least i dont see that on list of suported commands
https://github.com/gnea/grbl
must be the post processor you using that will insert that block.
is that a grbl post processor you use?
im using fusion 360 that have a grbl post processor sense is free for startups

also i usualy i dont use the M3 S3000 on that block sense then soon you stat a job spindle will turn on and probably will drag along the board and scratch, i usualy put S1000 M3  on the line when the machine goes to position on where it will start job, do when you press start machine goes up , goes to the position it will start and then turns spindle on and starts job, also do same on end of code with S1000 M5
 i dont  know how others do but thats how i do

allcustomz, 2017-06-27:
@vmeph @winder  thank you i have to read again but im have issues again its doing it with all file again i never had it like this but it starts the error then skip then it says "alarm 2 reset to continue" but i cant reset but every gcode file dose now i go back delete but still same?? Also one file i was trying ran half wat then stoped saying alarm 2 reset again but its just weird? Anything i can try

vMeph, 2017-06-27:
do you have soft limits and hard limits enable on config?

allcustomz, 2017-06-27:
@vmeph soft limits $1 hard limit 0$

vMeph, 2017-06-27:
The file u sending is bigger then the machine space so will issue alarme before eaven start i think, put soft limits at 0 and try same file

vMeph, 2017-06-27:
so lets say on the $ max travel mm for X you have 100 and you have soft limits at 1 so if you send a file lets say with X101 the machine when you hit start will not eaven start it will send you a alarm  cause you file X101 is bigger then the max travel mm

allcustomz, 2017-06-27:
@vmeph ok ill try soft=0 and hard=0 right?  also i work with f-engrave and size is good in file but i think its u.g.s because all my old f-engrave files that ran great every time now dont run

vMeph, 2017-06-27:
[image]

alarm 2 read what says

vMeph, 2017-06-27:
thats actually a good thing that will alarm before eaven starts, that way you know something is wrong with the gcode, that way you dont eaven start job and waste material after realize that the code size was bigger then the machine

allcustomz, 2017-06-27:
@vmeph  well the only problem is my old gcode files do the same and i have cut them alot of time before also thank you im trying now i appreciate all your help.   But its weird all the files are in the size perimeters? Weird

vMeph, 2017-06-27:
@allcustomz good question not sure what is wrong with your files or in what is happening
last file you post of your code  i seen you use in inch (G20)
i use my stuff in mm(G21) and i can see your file was 251.72 mm on the X, no idea if that is bigger then your machine max travel mm
but they should work if you put the soft limits at 0?
you can make a test to see if it works?

allcustomz, 2017-06-27:
@vmeph its running now it hasn't stopped yet but im not getting my hopes up yet. Thank you so much for all the help. So soft limits stopes it from going to far?? Also why outta noware? My routers frimware all the numbers we correct as in my settings? Wonder why it stoped working? Thanks though

vMeph, 2017-06-27:
So what happens is on settings $130,$131,$132 you define the max settings for X,Y,Z travel,so when you enable the soft limits to1,it will be referece to your max travel settings, so any time you make a gcode file that will go more the the max travel settings it willnot run the file and will issue a Alarm,witch is a good thing,that way will not let you eavean run the file cause the job size is bigger than your machine and otherwise it would make you waste material.

allcustomz, 2017-06-27:
@vmeph great your great also i switched my soft limit back to 1 after to see if it would still run and is cutting now great! Thank you for replying and all the help its really appreciated tgank you

vMeph, 2017-06-27:
Now sense you in front of your machine you are the only one that can see what gcode files you have and what settings you have set to see exactly what was the problem that most like it has to do with what we have talk about, also see always if you working in inch or metric cause inch bigger then mm and can make the diference,if any more questions dont hesitate to ask we here to help etch other,is my pleasure help out has long ohave answer for you.
Most comum problems here usualy related to noise issues, but i dont think this is the case
Sorry my english :)

allcustomz, 2017-06-27:
@vmeph yea and i agree. Also this is a great place thanks so much 😁