winder/Universal-G-Code-Sender #681: grbl has not finished booting

andrewopenshaw, 2017-06-29:
Hi Will,

I am looking for some help.

I have bought the OX cnc from ooznest and i am having difficulty in connecting it to my laptop.
i have downloaded the universal G code sender v1.0.9 and the windows driver from ftdichip.com/drivers/vcp.htm.  Baud rate set to 115200 and firmware is set to GRBL, but i only get the message "grbl has not finished booting".   I can normally sort most things out, but this has got me stumped.  can you offer any advice.

thanks

Andrew

winder, 2017-06-30:
This is usually because you're using the wrong baud rate or haven't flashed GRBL properly. If you have an old version of grbl you could try the old default of 9600.

andrewopenshaw, 2017-07-02:
Hi Will,

Can you explain what you mean by 'haven't flashed GRBL correctly'.  I have searched for the meaning, but can find very little.

I have tried both the 1.0.9 UGS build and the 2.0, both with the same result.
I have deleted and reinstalled the ftdichip drivers and confirmed that the baud rate is set to 115200, this has not helped either.

thanks

Andrew

winder, 2017-07-04:
@andrewopenshaw UGS opens the serial port provided and waits for GRBL to send a status string. Until the status string is seen you'll see this message.

Are you using the correct serial port?

closed 2017-07-09 by winder