winder/Universal-G-Code-Sender #683: sometimes homing not supported

peezal, 2017-07-05:
I have been going nuts. New at this and built bobs E3. Works great. Using universal G code 2.0 nightly build. It connects to the cnc and more often than not it says homing not supported. Strange as a few times it has homed and I could move router. Now im back to not supported. any help would be appreciated,

vMeph, 2017-07-05:
do you have homing enable in grbl sets?

peezal, 2017-07-05:
When I try to check in fermware settings it wont allow me and states grbl has not finished booting. even 30 minutes later it says the same thing.

From: Meph1 
Sent: Wednesday, July 05, 2017 9:59 AM
To: winder/Universal-G-Code-Sender 
Cc: peezal ; Author 
Subject: Re: [winder/Universal-G-Code-Sender] sometimes homing not supported (#683)

do you have homing enable in grbl sets?

peezal, 2017-07-05:
baud rate is set at 115200. grbl wont finish downloading.

From: Meph1 
Sent: Wednesday, July 05, 2017 9:59 AM
To: winder/Universal-G-Code-Sender 
Cc: peezal ; Author 
Subject: Re: [winder/Universal-G-Code-Sender] sometimes homing not supported (#683)

do you have homing enable in grbl sets?

winder, 2017-07-05:
@peezal if grbl doesn't connect after a couple seconds it probably isn't going to.
1. Check the baud rate is 115200, or 9600 for very old versions of grbl.
2. Make sure you are connecting to the correct port (and have any required drivers installed).
3. Make sure GRBL is properly flashed on your controller board.

peezal, 2017-07-05:
am at 115200 and correct comport. 
currently trying to flash grbl. Having a few issues with that, looking at many videos however nothing goes as smooth as the video. 

From: Will Winder 
Sent: Wednesday, July 05, 2017 4:44 PM
To: winder/Universal-G-Code-Sender 
Cc: peezal ; Mention 
Subject: Re: [winder/Universal-G-Code-Sender] sometimes homing not supported (#683)

@peezal if grbl doesn't connect after a couple seconds it probably isn't going to.

  1.. Check the baud rate is 115200, or 9600 for very old versions of grbl. 
  2.. Make sure you are connecting to the correct port (and have any required drivers installed). 
  3.. Make sure GRBL is properly flashed on your controller board.

peezal, 2017-07-05:
By the way, I appreciate your help.
many thanks

From: Will Winder 
Sent: Wednesday, July 05, 2017 4:44 PM
To: winder/Universal-G-Code-Sender 
Cc: peezal ; Mention 
Subject: Re: [winder/Universal-G-Code-Sender] sometimes homing not supported (#683)

@peezal if grbl doesn't connect after a couple seconds it probably isn't going to.

  1.. Check the baud rate is 115200, or 9600 for very old versions of grbl. 
  2.. Make sure you are connecting to the correct port (and have any required drivers installed). 
  3.. Make sure GRBL is properly flashed on your controller board.

winder, 2017-07-06:
Happy to help.

If the video's aren't complete you can try the grbl wiki page which has a lot of detail for all operating systems: https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino

winder, 2017-07-09:
Feel free to keep asking questions if you have more problems. I've started a troubleshooting page so in the future it will be easier to answer these sorts of questions.

http://winder.github.io/ugs_website/guide/troubleshooting/

closed 2017-07-09 by winder

peezal, 2017-07-09:
I had to set it aside for a few days. Just can not get it to work. very frustrated. currently trying to see if grbl is loaded on board but getting board not responding.

From: Will Winder 
Sent: Sunday, July 09, 2017 11:00 AM
To: winder/Universal-G-Code-Sender 
Cc: peezal ; Mention 
Subject: Re: [winder/Universal-G-Code-Sender] sometimes homing not supported (#683)

Feel free to keep asking questions if you have more problems. I've started a troubleshooting page so in the future it will be easier to answer these sorts of questions.

http://winder.github.io/ugs_website/guide/troubleshooting/

vMeph, 2017-07-09:
did you  upload correctly the grbl to the arduino?
to upload correctly from ide arduino you cant have old grbl folders in the libaries

peezal, 2017-07-30:
reflashed grbl and it works. CNC will home.

vMeph, 2017-07-30:
Cool