winder/Universal-G-Code-Sender #687: Z Height Probing?

aforww, 2017-07-11:
Ok, maybe it's my lack of sleep and the heat is getting to me but I can't get probing to work at all and kind find any documentation on it. I've got Pin A5 and Ground and it will start, then stop after about a mm of travel and I have to close UGS. It says "clear alarm" in the controller state. Nor do I know what is supposed to happen when it completes telling if it worked. Obviously, it needs to adjust the code for the offset but how do I know it was successful and accurate?

vMeph, 2017-07-11:
it should be something like this
Settings

-probe feedrate =set feedrate(velocity) you want the proble to move

-outch plate thickness=the thickness of the base plate you using in mm or inch depending how you got set your system

-probe distance = the distance that Z  will travel until toutches the plate, got to travel more then the distance to reatch the plate or it will stop(ex: if to toutch plate you got to travel Z0 to Z-10 you set like Z-Z-12 or more

-probe axis you put Z

connect the  A5 to the plate with a corcodile clamp
connect ground  to the drill bit with a cocodile clamp

- hit iniciate button to start probing

what you expect to happen is the z will start moving down until toutches the plate and will stop when it toutches plate

after probe when you send a file to start a job it should offset the Z coords of file

if alarm happens it could be cause or the distance of travel is not enought to toutch plate 
or you have noise issues on your lines

aforww, 2017-07-11:
@vMeph, I've got all that correct (thankfully). However, I can't see that it's issuing any command to set the offset. 
I initiate probe. It does it's thing, stops, and issues "G90 G21". I was expecting G90 Z-12.7.

aforww, 2017-07-11:
Sorry, meant I expect something like G92 Z-12.7

vMeph, 2017-07-11:
what i get when hit iniciate is
>>> G21 G91 G49
>>> G38.2 Z-10 F25
>>> G90 G21
ok
final out put
[PRB:0.000,0.000,-5.138:1]
ok
ok
>>> G21 G90
>>> G43.1 Z-7.138
>>> G90 G21
ok
ok
ok

Im using UGS platform

aforww, 2017-07-11:
Hmmmm. I'm using Platform. 2.0 with GRBL 1.1f. This is what I get. My plate is 12.7mm thick. 

k
>>> G38.2 Z-20 F50
[PRB:-248.000,-263.000,-30.200:1]
ok
>>> G90 G21
ok

aforww, 2017-07-12:
Ok, It is working now. Sort of. Sometimes it issues the 43.1. Sometimes it doesn't.

winder, 2017-07-15:
@aforww let me know if you can figure out a pattern for when it doesn't work.

aforww, 2017-07-24:
Well now it's consistently not automatically compensating for thickness of
touch plate. Hasn't done it since I posted this lol.

aforww, 2017-07-24:
@winder my process is, home machine, jog to approximate WCS zero, run probe helper. It runs the probe command but never issues the 43.1.

winder, 2017-09-02:
@aforww I finally got end-stops and a probe installed on my machine and gave this feature a real try. Sure enough, it didn't work at all with homing enabled. This machine is using grbl 0.9, but I don't think that should make a difference for the changes I needed to make.

I spent some time this evening fixing it.  When you have a chance please give this feature another try and let me know if it's working now.

aforww, 2017-09-02:
Man I totally forgot about this. I haven't even turned on my machine since this. We just got power back (live in the Hurricane Harvey path) and got the message cleaned up so I'll update and fire up the machine today to give it a go. Awesome job following up on it though. I'll get back to you today.

aforww, 2017-09-02:
*mess

aforww, 2017-09-05:
@winder good news, after a few attempts it seems to be behaving as it should. I'll update if I run into any issues. Awesome job man.

closed 2017-11-07 by winder