winder/Universal-G-Code-Sender #689: Stepper Driver Losing Power - Stalling

Levi79, 2017-07-15:
Hello I'm new to CNC and have been having a really hard time with this issue while using the most recent UGS Platform. For some reason when I create gcode from Fusion 360 it seems to be sent without proper acceleration on G1 moves, this keeps causing one of my two Y stepper drivers to lose power and stall. All of the codes I have run from the previous owner of the machine (using HSMWorks) have gone flawlessly. Wasting bits and material when it stalls is no fun so any help or ideas would be much appreciated!

Thanks
Levi

winder, 2017-07-15:
Acceleration is controlled by the GRBL firmware, the CAM software and UGS can only provide a value for the target feed rate with the F command. The F code is a global state setting, so it doesn't need to be provided with every command.

Have you reconfigured GRBL since acquiring your machine?

jahnj0584, 2017-07-15:
Will, I'm the ex-owner. It's a real puzzle of an issue, since this issue
happens with moves that are only at 600mm/min while it can happily jog at
5x that. It's not an issue dependant on location of the machine either
(since my codes run fine multiple times)

Levi79, 2017-07-15:
Yeah I've gone over the GRBL settings more than once to make sure everything looks correct. As Julius said there is no issue with jogging the machine at high speeds or running his gcode so we're just not sure why it isn't getting along with my Fusion 360 generated code.

winder, 2017-08-30:
Sorry but I'm not sure whats going on here. If you haven't already you could try sharing the two files (HSMWorks vs. Fusion360) on the GRBL issue board. There have also been some issues with the grbl fusion 360 post processor, there are several 3rd party post processors you could try.

closed 2017-08-30 by winder