winder/Universal-G-Code-Sender #690: NullPointerException while parsing GCode file

tkkuehn, 2017-07-17:
Hi,

I'm running the July 16, 2017 nightly build on Arch Linux.

When I attempt to parse any GCode file (I believe each one I've tried has been valid) from the `Browse` dialog, the file fails to parse and the logs indicate a NullPointerException.

Sample gcode file:
[sample_gcode.txt](https://github.com/winder/Universal-G-Code-Sender/files/1153878/sample_gcode.txt)

Sample output:
```
SEVERE: Problem while browsing.
java.lang.NullPointerException
	at javax.vecmath.Tuple3d.<init>(Tuple3d.java:91)
	at javax.vecmath.Point3d.<init>(Point3d.java:74)
	at com.willwinder.universalgcodesender.types.PointSegment.<init>(PointSegment.java:82)
	at com.willwinder.universalgcodesender.gcode.GcodeParser.addLinearPointSegment(GcodeParser.java:232)
	at com.willwinder.universalgcodesender.gcode.GcodeParser.handleGCode(GcodeParser.java:312)
	at com.willwinder.universalgcodesender.gcode.GcodeParser.processCommand(GcodeParser.java:210)
	at com.willwinder.universalgcodesender.gcode.GcodeParser.testState(GcodeParser.java:438)
	at com.willwinder.universalgcodesender.gcode.GcodeParser.preprocessCommand(GcodeParser.java:421)
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.write(GcodeParserUtils.java:71)
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.processAndExportText(GcodeParserUtils.java:124)
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.processAndExport(GcodeParserUtils.java:54)
	at com.willwinder.universalgcodesender.model.GUIBackend.preprocessAndExportToFile(GUIBackend.java:163)
	at com.willwinder.universalgcodesender.model.GUIBackend.initializeProcessedLines(GUIBackend.java:909)
	at com.willwinder.universalgcodesender.model.GUIBackend.setGcodeFile(GUIBackend.java:476)
	at com.willwinder.universalgcodesender.MainWindow.browseButtonActionPerformed(MainWindow.java:1271)
	at com.willwinder.universalgcodesender.MainWindow.access$3000(MainWindow.java:84)
	at com.willwinder.universalgcodesender.MainWindow$28.actionPerformed(MainWindow.java:993)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at com.willwinder.universalgcodesender.utils.GUIHelpers.lambda$displayErrorDialog$0(GUIHelpers.java:45)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

winder, 2017-08-10:
Sorry for the delay.

The problem is that UGS isn't handling the `G0` with no coordinates. I thought for sure this wasn't valid or not, but looking at [the standard which GRBL follows](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g0) it is:

>     For rapid motion, program G0 axes, where all the axis words are optional.

Thanks for reporting this edge case!

winder, 2017-08-10:
This bug is now resolved in the latest nightly build.

commit d81bb035 (2017-08-10): Allow linear motion commands with no axis words. Closes #690
closed 2017-08-10 by winder