winder/Universal-G-Code-Sender #691: UGS platform $C

vMeph, 2017-07-17:
@winder one question when UGS uses the comand $C to view code with out moving machine if it find a error a pop up comes asking if you want to disable comand on future or no, but if you say no then you cant see the gcode no more passing 
is that normal?

vMeph, 2017-07-20:
@winder
Was asking this cause i was cheking a file that as alot errors and the $C only shows the first one and stops with ok ok ok
Is this how is supose to act or should report all errors 1 at time?

winder, 2017-09-04:
Technically this is the correct behavior, "Check Mode" is a grbl idea and UGS sends code as it normally does without knowing that grbl isn't actually moving the machine.

The behavior could probably be improved to generate an error report instead of pausing on errors like it would normally.

winder, 2017-10-13:
I'll fix this as part of #759

closed 2017-10-13 by winder