winder/Universal-G-Code-Sender #693: Fusion 360 post to ugs

allcustomz, 2017-07-21:
[image]

allcustomz, 2017-07-21:
Ware did i go wrong im trying to post from fusion 360 to ugs i get the add in and then it pops up what to post and ware. But i get this error any ideas?

vMeph, 2017-07-21:
Hmm you dont post directly to ugs you post on brackets save the code in a file.  .nc and then open from ugs the file

allcustomz, 2017-07-21:
i have saved file but ugs won't read it because the code is different is there anything else I could do or something I need to turn on or install I installed grlb and grlb plus then installed a post prossesser from  a guy at fusion for ugs but i get this error. Did i paste fine wrong when you add add_ins? Or how do you do it? Before all this ugs wouldnt read file? Thanks for the reply im stuck

vMeph, 2017-07-22:
did you select on post processor the grbl.cps has the post process output?
pic of your post process settings?

allcustomz, 2017-07-22:
@vmeph thanks i think i didnt need it just turn of g24 i think and send file it was great thanks it was overkill putting new post prossesser when i didnt need it thank you again for your help im cutting now with no problems thanks your a lifesaver

vMeph, 2017-07-22:
You welcome mate, you can close post

closed 2017-07-22 by allcustomz