winder/Universal-G-Code-Sender #696: UGS Reading NC not NGC files

CdoubleOkay, 2017-07-25:
I recently updated my nightly build and am trying to run laser files. When I go to load the file, the file name shows but no rows in the file. The laser files are NGC files. When I load up a carving file (NC) it comes up and runs fine. I even changed the file extension of the laser file to a NC and it still does not work. I am running GRBL 1.1 and have it on Windows XP with Java 8 update 141 build 1.8.0_141. I did get an error with F100 command. Drawing a complete blank here and have a project to finish by Thursday

CdoubleOkay, 2017-07-25:
Here is one of the files I am trying to run. UGS reads the G20 command and the F100 but doesn't process anything else
[MhA 1.5 Inch.zip](https://github.com/winder/Universal-G-Code-Sender/files/1174564/MhA.1.5.Inch.zip)

CdoubleOkay, 2017-08-10:
@winder any chance to have a look at this?

winder, 2017-08-10:
Is this what you expect the program to look like?
[image]

You might have accidentally misconfigured the gcode parser, take a look at this document for how to fix it: http://winder.github.io/ugs_website/guide/troubleshooting/#gcode-program-stopped-working

If that doesn't work I'll need to ask a few more questions about how things are failing.

CdoubleOkay, 2017-08-11:
@winder Yes, it's a laser engrave however when I use the 2.0 Classic it does not read the file and returns a null stream

closed 2017-08-11 by CdoubleOkay