winder/Universal-G-Code-Sender #703: UGS dimensions problem

scottcpledger, 2017-08-07:
I just started having an issue out of the blue today with my cnc machine.   When I load a file the correct dimensions show up in the visualizer , although they are in metric, but the correct dimensions.  When I connect my cnc machine to UGS the visualizer changes the dimensions to standard measurements (inches) but the dimensions are wrong and much much smaller than what the real dimensions should be.   For example: before connecting my machine dimensions show 88mm H x 88mm W then when I connect my cnc machine the dimensions change to .14" H x .14" W.    basically unable to use my machine because of this.   Like I said it just started happening. I even downloaded a different version of UGS hoping it would fix with no luck.

jahnj0584, 2017-08-07:
Did you try reloading the file or only opening it when your already
connected?

scottcpledger, 2017-08-07:
Thanks for the reply. Yes I tried several files that I use all the time.  I
also tried recreating some and same thing.

winder, 2017-08-07:
@scottcpledger are you using the latest nightly or stable build of 2.0?

scottcpledger, 2017-08-07:
Yes 2.0 nightly

winder, 2017-08-07:
@scottcpledger seems like I've got a units problem. This should only be a display issue in the visualizer though, if you wanted to run the file I believe it should be just fine. Regardless I need to fix the display problem. I ran a test with `$13=1` and a gcode file using inches and it seems like the conversion is correct - one axis goes from `193.19mm` -> `7.61"` after connecting.

Could you share the file you're using?

What is the output of `$$` for your GRBL settings?

What version of GRBL are you using?

winder, 2017-08-07:
This turned out to be a regular expression pattern issue:
https://discuss.inventables.com/t/universal-gcode-sender-problems/40518/9

#680 will hopefully make that problem easier to recognize.

closed 2017-08-07 by winder