winder/Universal-G-Code-Sender #704: Unable to Autoload Bluetooth as Serial Port

Lizichka, 2017-08-13:
I've been using UGS for a while now and it's great. There is one issue that keep bugging me. 

I'm using a Bluetooth to send Gcode to my CNC. I have to go to terminal to locate the name of the Bluetooth and then type it into UGS connection port manually ( ls /dev/tty.arduimu-iface-SPP, for example). I'm using Mac OS 10 if that matters. 

Is it possible to auto-load Bluetooth as a serial port on UGS in connection section - port down menu? 

It would be a time saver and greatly appreciated. Thanks.

pelrun, 2017-08-14:
UGS uses the JSSC library to get the serial ports for a platform, so it'd be more effective to submit the issue over there: https://github.com/scream3r/java-simple-serial-connector

winder, 2017-08-14:
Unfortunately, JSSC hasn't been updated in a long time. The Arduino-IDE has worked around this by maintaining a fork of [SerialPortList](https://github.com/arduino/Arduino/blob/ec2e9a642a085b32701cf81297ee7c1570177195/arduino-core/src/processing/app/SerialPortList.java). I'm reluctant to take on the additional maintenance with all the other things I want to get to on UGS.

closed 2017-08-14 by winder