winder/Universal-G-Code-Sender #706: Program Window is Very Small

cuttergcode, 2017-08-18:
When using UGS on my computer the window is very small.  It's very hard to read the words and buttons.

Microsoft Surface Book
Windows 10 Pro
3000 x 2000 screen resolution

Is there a way to make it normal like my other programs?

winder, 2017-08-18:
Currently, your options are to install an early release version of Java9 (which has HiDpi support and is due for a proper release later this year) or use the platform version of UGS which uses a framework which supports this.

Related to #623

closed 2017-08-18 by winder

Lizichka, 2017-08-20:
I'm having the same issue. Can you link to th early release of Java9 that supports HiDpi please? Can't find it. Would I have uninstall java from my computer?