winder/Universal-G-Code-Sender #707: toll does not move

salimOfTizi, 2017-08-19:
Hi every one. First, excuse me for my bad English.
i  have a problem with the visualize window of UGS : the tool does not move when i send the G-code file

winder, 2017-08-20:
What version of GRBL and UGS are you using?

salimOfTizi, 2017-08-21:
I'm using GRBL 0.9j   and  UGS  1.09 .
Ihave  a lenovo thinkpad with window7 64 bits installed.

winder, 2017-08-28:
When you say the tool does not move, do you mean in UGS, or on your machine?

If it is in UGS you might need to fix the the `$10` setting to `$10=3` ([see document here](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#10---status-report-maskbinary)).

If your machine doesn't move you might have it wired wrong, you can check that here: https://github.com/grbl/grbl/wiki/Connecting-Grbl

salimOfTizi, 2017-08-28:
It works !!!.Thank you very very very.......very much .You are a genius!!!!

closed 2017-08-28 by salimOfTizi