winder/Universal-G-Code-Sender #708: Blacklisted commands is not intuitive for new users  (regular expression pattern remover)

rlwoodjr, 2017-08-20:
70% of my emails to help users with software with issues are related to the blacklisted commands.  Can we disable this feature as a default? (regular expression pattern remover).  It would appear folks are just clicking yes, 

For example, sometimes g20 gets on the list, Then UGS displays the g-code in the default mm units. 

Another option is to take them to the "controller option" screen so the operator can know when a command is being blacklisted and where it is.

winder, 2017-08-20:
Disabling the feature until I am able to finish #680 is probably a good idea

winder, 2017-10-13:
@rlwoodjr Finally got around to disabling this, sorry for the delay.

commit 5f46fe60 (2017-10-13): Disable 'ignoreFutureErrors' dialog, improve error detection at beginning of file, improved error message in console. Closes #708
closed 2017-10-13 by winder