winder/Universal-G-Code-Sender #709: Keymap editor doesn't recognize UP or DN arrow keys

groberts22, 2017-08-20:
This is my first time attempting to submit an issue.  I hope I am doing it correctly.  I have been using UGS for some time and love it.  Thanks for all the hard work.  I recently downloaded the 7/23 nightly build of the UGS Platform and am having trouble with the keymap editor.  Specifically, it won't let me map jog commands to the UP and DN arrow keys, as I have done in the past.  Thanks in advance for any help you can provide.

jahnj0584, 2017-08-20:
I think i had that issue too. Simply add a "trigger" before and you can use
the arrow keys. I used alt, so you HAVE to want to jog it. Regular arrow
keys could be mistakingly hit while browsing the sent commands or whatever.

groberts22, 2017-08-20:
The trigger keys won't work for me, as I am trying to use a wireless keypad that doesn't have the triggers.  All of the other keys work, including the LT and RT arrow keys.

jahnj0584, 2017-08-20:
No control or shift?

groberts22, 2017-08-20:
Yes.  That is correct.

groberts22, 2017-08-21:
It looks like the problem may occur because the UP and DN arrow keys are used to scroll through the recommended free shortcuts in the keymap editor.  This is really unfortunate, as these are natural keys to use as jogging commands.  Is there any way this can be fixed, or is there a work around?  The wireless key pad otherwise works great as a "poor man's" pendant.

vMeph, 2017-08-21:
I woul love to see this fix to happen
@winder any way to make arrows keypad keys to work with out having a trigger key selected?

vMeph, 2017-08-21:
Also if there was a way to add time ms selection for key response would improve jog response,i recall try with a joystic and a program that add a option to input time response and the jog worked so perfect, what i mean is when you jog with keys is not imediate response to start and to stop when you release key,specialy when is to stop you release key and doest stop right way always stop further.

groberts22, 2017-08-21:
I think the most recent release of grbl implements jog capability with pretty low latency, although also offers an adjustment capability.

winder, 2017-08-22:
I was generally aware that this wasn't handled very well in the platform. There are two separate problems I'm aware of:
1) The keybinding window doesn't let you assign UP / DOWN because (as @groberts22 noticed), that triggers a recommendation UI element.

2) Some modules take over common keys and prevent key bindings from being triggered. For example, if you start typing in the command console characters will be typed. So anything bound to "1" will be added to the text widget instead of triggering a keybinding.

I think the second issue is the bigger problem for jogging. On my Mac at least, if I have the cursor in a text box even COMMAND-ARROW doesn't work. If I deselect the text area (clicking a button or closing a window), then keybinds like "RIGHT" work with no modifier.

I'm not sure I can solve the first issue. If you're persistent you can find the text file where key bindings are stored and manually change it -- but that isn't a real solution.

What I would like to do is have a special "JOG MODE" button in the toolbar which would create a modal dialog. Jogging would be disabled unless that dialog was open, this way there would be no chance for accidental jogs (or ignored jogs). Something like this is what I have in mind where the text boxes let you reconfigure (or see) the bindings right away:
[image]

This would also allow me to implement more advanced jogging where I do things like start jogging on KEY_PRESSED and stop on KEY_RELEASED and use the [joystick implementation described here](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging#joystick-implementation).

winder, 2017-08-22:
Maybe it doesn't have to be a modal dialog, that could obscure the DRO window. Maybe I could change the background from the default gray color to green when the window is enabled as an indicator.

groberts22, 2017-08-22:
Thanks for the thoughtful response. I wish my understanding of the underlying program structure was better, but alas, I am a relative newbie and still learning.  I like the idea of the dialog box.  For what it's worth, the layout of the wireless keypad I am using suggests a couple of changes that might simplify the layout and provide useful feedback regarding the current state of the Jog Controller.  Would it be possible to display the current distance and feed rate settings and then use the "/", "*", "-", and "+" keys to alter them?  Another key could then be used to toggle the modifier keys between the distance and feed rate settings.  My keypad has a "backspace" key below the "+" key that would work well for this purpose.  Similarly, the "Home" or "End" key could be used to toggle between mm and in mode.
I would mock this up if I knew how, but unfortunately my skills are still in "beta".  In any event, I really do think some such implementation could be of great value to a lot of users like me who don't have their computers located next to their machines.

winder, 2017-08-22:
@groberts22 could you try using this to mock something up? It sounds like you have some good ideas about how this might work: http://gluonhq.com/products/scene-builder

groberts22, 2017-08-22:
I'll give it a shot.

jahnj0584, 2017-08-22:
I'm still in support of just using a trigger key to prevent accidents. You
could drop a screw or clamp or the keyboard itself and ruin a bit and or
your work piece because of not having one.

Alternatively, Winder, do something like a Haas controller. You must be in
the MDI or manual mode "windows" to hand jog the machine. Maybe just a
simple shortcut that enables jogging as a toggle.

groberts22, 2017-08-22:
I took a shot at mocking up my vision of the jog control interface.

The first image shows a prototype of the jog control window.
1. The 6 buttons on the left work as expected - pushing any one of them jogs the tool one step in the specified direction at the indicated feed rate. [It might be possible to program these keys so that they work like momentary switches per @winder's previous suggestion.
2. The Units button toggles between inches and millimeters, and displays the current setting.
3. The six boxes to the immediate right of the Units button display the current step size and feed rate values for the X, Y, and Z axis. It might be possible to program the interface so that these values can be entered manually: alternatively (or in addition) they could be set using the 5 buttons on the far right.  My thought is that the Axis/Step/Feed button could be used to toggle between the six windows and the values could then be adjusted using the 4 remaining buttons.

[image]

The next image shows the key assignments used to implement the functionality of the jog controller on the wireless keypad.  I chose the keys based on the layout of the wireless keypad I am currently using.

[image]

The final image shows the keypad itself, which I bought on Amazon for $10.  I have been very happy with it.  As you can see, I have used most but not all of the keys on the keypad to implement the various jog control functions.  It might be useful to map the unused Home, End, and Back Space keys to other functions that frequently come in handy while standing at the machine (e.g., to initialize a homing sequence or a z-axis zero setting macro after a tool change).   I am not sure about the 5 and Enter keys, although in principle they could be mapped as well.  And while I am dreaming, note that the Num Lock option provides the potential for many additional mappings.

[image]

vMeph, 2017-08-22:
Very nice job, i also have a keypad like that, i like it all with the exeption of The Units button toggles between inches and millimeters, in my case i dont see to mutch use for it, sense im in europe i use mm, never have used inch in my machine

groberts22, 2017-08-22:
If only! Sooo many things in the US are specified in inches.

jahnj0584, 2017-08-22:
Units should be a preference menu option. There's no need for it to be
changed after the initial machine set up

groberts22, 2017-08-22:
That makes sense to me. I included the option without thinking too much about it, because it was in the existing prototype.

winder, 2017-10-13:
superseded by #779

closed 2017-10-13 by winder