winder/Universal-G-Code-Sender #710: Issues with high feed rates

jahnj0584, 2017-08-22:
Testing my motors before I mount them. It seems that if I put the feed over 12500mm/min or so the red DRO value freezes, the stop command does nothing, and my motor shaft locks in place. The alarm light does NOT go off on my driver. 

200 steps/mm
15000 Max feed 
250mm/s2 acceleration 

Does this have something to do with the driver's 20kHz Max pulse rate or the arduinos computational power?

pelrun, 2017-08-23:
12500/60 * 200 = 41kHz, so yes.

jahnj0584, 2017-08-23:
Weird then that 12400 will send. Does it have anything to do with the pulse
time of 10 or is that my limit?

I guess it's a first world problem lol.

winder, 2017-08-28:
@jahnj0584 when you say the red DRO value freezes does that mean the number is red and doesn't change? I'm trying to understand if there's a GUI bug in addition to trying to running the driver too fast.

jahnj0584, 2017-08-28:
yes exactly, the DRO freezes at its value and stays red. It stays as such
even if you wait for the motor to come back. It requires me to force quit
UGS and restart it.

MattyJ4, 2017-09-01:
I am experiencing the same issue. 

My steps/mm on $102=39500 (due to non adjustable micro stepping on my driver) 

When I try to increase my max on $112 everything stops and needs a hard reset.

inFamousMcGrath, 2017-09-19:
GRBL really can do 25-30kHz max in ideal conditions. You must use 32bit platforms, like Smoothieboard, or ported GRBL version on STM32 microcontroller.

PR #1113 (2018-12-31): Feature/setup wizard rate warning
Fixes a bug where the steps/mm-setting wasn't updating. 
Also fixes #710 informing the user that their setting is exceeding the machine limits:

[image]
closed 2018-12-31 by winder