winder/Universal-G-Code-Sender #713: Home position and stepper direction won't add up

Nube123, 2017-08-23:
Trying to set up my home made CNC router using a grbl flashed "arduino".
I want my home position to be in the down left corner, the X+ direction to be to the right hand Y+ to be upwards.
As it is now, home is in the upper right corner and the X and Y axis moves opposite of the selected direction. 
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=1200.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$130=500.000 (x max travel, mm)
$131=500.000 (y max travel, mm)
$132=150.000 (z max travel, mm)

The Z axis is ok and i have been trying to change the cable direction on the steppers, inverting the directions and inverting the homing dir($23). 

When i inverted the homing dir the home position got an offset negative to the max travel i have set
$130=500.000 (x max travel, mm)
$131=500.000 (y max travel, mm)

vMeph, 2017-08-23:
The $23 last  maksk 3 bits 000 is equal to xyz so if you want to invert the direction of homing you got to set to 1 the axis you want to invert,exmple if u wanted to set the xy to invert direction you must mask last bits to 110 so the $23=6, cause 110 binary in decimal is 6,just dont remember if the order of last 3bits on axis order is correct 000, xyz, but i think thats it

Nube123, 2017-08-23:
I tried it but it didnt work. in the configuration it seems that $23=6 is inverting y and z, i've also tried to $23=3 that inverts the x and y axis.
Setting Value | Mask | Invert X | Invert Y | Invert Z
-- | -- | -- | -- | --
0 | 00000000 | N | N | N
1 | 00000001 | Y | N | N
2 | 00000010 | N | Y | N
3 | 00000011 | Y | Y | N
4 | 00000100 | N | N | Y
5 | 00000101 | Y | N | Y
6 | 00000110 | N | Y | Y
7 | 00000111 | Y | Y | Y

vMeph, 2017-08-23:
What cnc shield and grbl version you using

vMeph, 2017-08-23:
Apart from it CNCs not like 3d printers they work on negative space thats why then you use work coordinates to set 0,0,0 work space

Nube123, 2017-08-23:
I flached a arduino uno(fake) using x-loader with v1.1 (grbl-master)

vMeph, 2017-08-23:
What Shield you using?

Nube123, 2017-08-23:
using the arduino as shield: https://github.com/grbl/grbl/wiki/Connecting-Grbl
with TB6560 3A CNC Router

winder, 2017-08-27:
As mentioned by others you need to configure `$23` in GRBL, in the table you posted it seems that `$23=3` is what you need to invert the XY axes and put the homing position in the lower left corner of the XY plane.

If that isn't working you might need to post a question on the GRBL issue page.

closed 2017-08-27 by winder