winder/Universal-G-Code-Sender #716: Limited step size

jahnj0584, 2017-08-23:
https://youtu.be/0XpA5HUVdb8

pelrun, 2017-08-24:
The step size is either in mm or inches. Do you really need to jog in steps of ten metres?

jahnj0584, 2017-08-24:
My machine is only 2m so I understand this is a useless issue but it is
still one. Why have it limited?

pelrun, 2017-08-24:
If you can demonstrate an actual need for larger values than are currently supported, then that would warrant raising an issue. As it stands there is no need for this to be changed.

winder, 2017-09-04:
@jahnj0584 I don't think I added that limit intentionally, but I also don't think it's very useful to jog such a large distance. At some point, I'd like to revisit the jogging UI and this will hopefully no longer be relevant (i.e. press and hold the button instead of specifying the step size).

closed 2017-09-04 by winder

jahnj0584, 2017-09-11:
ok heres one. I have 1300mm of travel and need to find a spot that is binding. Two 650 commands wont work because you introduce the acceleration variable into the mix regards to motor velocity and steps/s instead have having a nice constant move.

pelrun, 2017-09-11:
Just use a manual gcode if you want something that specific!