winder/Universal-G-Code-Sender #717: UGS Platform Auto Updates

winder, 2017-08-26:
Over the past couple weeks I've been working to support an auto update feature with UGS Platform. There is still a bit of work to do, but the auto-update files are now being created during the nightly updates and uploaded to sourceforge where they seem to work. There are a few issues related to sourceforge that I'm still working out, but I think it works well enough that people can start trying it out.

Eventually this will be enabled by default, but for now you need to configure the "Update Center" manually. It is fairly straight forward:

Open the plugin menu:
[image]

In the settings tab add a new update center:
[image]

Fill in the settings...
Name: Nightly Build Updates
URL: http://master.dl.sourceforge.net/project/ugs-update-center/2.0-SNAPSHOT/updates.xml
[image]

The updates tab should now have a number of files to update, if it doesn't you can click "Check for Updates":
[image]

There isn't an easy way for me to enter a changelog for now. In the future, I may want to create a "whats new" type dialog. Also because the nightly build runs every night, it is likely that there will be updates available even if there were no code changes. In the future, I might have "Nightly Builds", "Stable Builds", and "Releases" as separate update centers which can be turned on or off depending on what type of releases you're interested in.

This is a feature I've been wanting to add for a very long time, if you give it a try please let me know how it goes!

rmisko11, 2017-08-27:
> 2.0-SNAPSHOT/updates.xml

Just set it up and so far so good.

vMeph, 2017-08-28:
@winder thanks alot for this implementation of auto updates this will make life easier for all
Would it be possible for this to be pre installed on the UGS software with out people have to set it up and folow steps, that could make life easier for new comers and less questions about it

Auto updates  is working like a charm!!!

winder, 2017-12-03:
Moved to google group: https://groups.google.com/forum/#!topic/universal-gcode-sender/GRK8HXwucaA

closed 2017-12-03 by winder