winder/Universal-G-Code-Sender #718: Jobs don't often end correctly

dmavrikakis, 2017-08-27:
I am having this issue quite frequently and it's becoming a real pain. After a job ends I find that often the Play and Stop buttons are active but everything else is grayed out. Pressing the stop button has no effect, if I disconnect and reconnect then things appear to be back to normal except that the jogging window will stay inactive until I close and reopen UGS. I am using the latest nightly build (UGS platform) and I also had the same issue with the stable builds (which is why I attempted the nightly build). My GRBL is 0.9. Any help would be very appreciated.

Macioza, 2017-08-27:
The same here. I'm having exactly the same issues with several past UGS builds. Updating GRBL has no effect (currently running 1.1f). Passing "$$" into command line, seems to help, but not always. I've experimented with different gcode endings, but that seems to have no effect whatsoever.

dmavrikakis, 2017-08-27:
Im glad to see that it's not just me having this issue, I was a little worried it could be a problem with my machine.

The only thing that I have noticed is that you can still press the Run button despite the jog controller being inactive and the job will start which means that its not the end of the world if you are doing lots of the same job. 

I have just cloned and got myself setup with the project so I will try running it in debug to see if I can track down the source of the issue and will report back with more info in the hope that we can get to the bottom of this.

winder, 2017-08-27:
@dmavrikakis this is probably something to do with the `AbstractController` having an extra command remaining in the `activeCommands` array.

There might be something about the program you're sending which is causing a problem with the UGS preprocessor which causes UGS to expect an "ok" that it shouldn't. Could you share a program that you're seeing this problem with?

Knowing that this problem happens with GRBL 0.9 and 1.1 is very helpful. Currently, 1.1 has better support (because most people reporting bugs are using that version) but it seems like the problem doesn't have anything to do with the version of GRBL.

dmavrikakis, 2017-08-29:
Rather frustratingly I can't repro it when debugging on my macbook connected to my cnc machine. The computer I actually use (a RPi) will repro it with out fail using a standard build, problem is I can't easily debug on the RPi.

When I get more time I will make a build with plenty of Logging in the AbstractController and run it on my RPi to see if I can identify why/where it's getting stuck. All I can say for certain at the moment is that I am often not getting the call to fileStreamComplete which is leaving the application in limbo.

I have attached a program, but I don't think you will find much in there, if so I would have expected to see the problem occur when running it with my macbook.

[AB.txt](https://github.com/winder/Universal-G-Code-Sender/files/1258749/AB.txt)

BusbyGT, 2017-08-29:
This is the same issue that I've reported in Issue #593

https://github.com/winder/Universal-G-Code-Sender/issues/593

I first saw it in v1.0c but it continues with v1.1.

I'm also using a Raspberry Pi as my main platform.

Macioza, 2017-08-31:
It's worth noting, that I'm also running UGS on Pi 2.

trbLeeciN, 2017-09-21:
I am also having the same issue for a long time. I tried to bypass the problem but it doesn't work.
I added G1X0Y0F1001 to the end of my gcode files. And I try to check if that command completed or not.
It sends G1X0Y0F1001, but it sometimes crashes when it comes to complete. Actually not only last row but also last 3rd or 2nd row are also missing too. What happens is, ugs thinks it is still sending commands, so it stays in that condition.

@override
public void commandSent(final GcodeCommand command) {
System.out.println("commandsent:"+command.getCommandString());

    java.awt.EventQueue.invokeLater(new Runnable() {
        @Override
        public void run() {
            // sent
            if (commandTableScrollPane.isEnabled()) {
                commandTable.addRow(command);
            }
            //commandTable.updateRow(command);
        }});
}

@Override
public void commandComplete(final GcodeCommand command) {
    
	System.out.println("commandcomplete:"+command.getCommandString());
    if(command.getCommandString().equals("G1X0Y0F1001")){
    	try {
    		Thread.sleep(100);
    		backend.sendGcodeCommand("$$");
    		Thread.sleep(100);
		} catch (Exception e) {
		}
    }
	
    // update gui
    java.awt.EventQueue.invokeLater(new Runnable() {
        @Override
        public void run() {
            if (commandTableScrollPane.isEnabled()) {
                commandTable.updateRow(command);
            }
            if (backend.isSendingFile()) {
                if (vw != null) {
                    vw.setCompletedCommandNumber(command.getCommandNumber());
                }
            }
        }});
    
}

trbLeeciN, 2017-11-28:
[image]
1- This happens sometimes, All files sent,done and got response OK, but somehow it can't finish stream. so it stucks like in the first image.

[image]
2- Later on, I click "Cancel" button. Only thing I notice is the change in duration value. 

[image]
3- I click "Cancel" button again, this time it enters "Hold" state from "Idle". (see active state value)

[image]
4- I click to "Resume" button, and finally UGS is ready to send file again. 

This is how I cope with this problem, I tried to solve it by checking duration time, if it stucks in 0 more than 2 seconds, I did this and that, but I couldn't fix it. I also tried using previous version of UGS,even in that version this problem occurs but less frequently. 
** By the way all my tries are in RPİ 3 environment.

breiler, 2018-03-26:
@dmavrikakis we have made some changes together with #759 that could resolve this issue as well. If you have the time I would really appreciate if you tried out the latest nightly build and verified that the issue is gone.

breiler, 2018-04-11:
I'm not able to reproduce the issue so I'm closing it.

closed 2018-04-11 by breiler