winder/Universal-G-Code-Sender #720: unknown error "wait for ok failed"

alihamza96, 2017-08-27:
my machine is not runing with a programmee and it is showing thistype of error, can any one help me
[image]

[image]

vMeph, 2017-08-27:
you asking suport for Grbl Controler program and this is program universal g code sender suport
what grbl version you using?

winder, 2017-08-27:
@alihamza96 My understanding is that the Grbl Controller project has been abandoned by the primary developer. If you want to try asking that community for support, this is where you would post the question: https://github.com/zapmaker/GrblHoming/issues

There are lots of active controller programs you could use. I can recommend UGS (of course), or you could explore [some of the many alternatives](https://github.com/gnea/grbl/wiki/Using-Grbl).

closed 2017-08-27 by winder