winder/Universal-G-Code-Sender #721: need 4 axis to drive 2 Y axis motor drivers

Sunnysky, 2017-08-27:
Will it be  possible to include a 4th Axis?

I have a 1.2m square gantry with 2 Y axis  motors & drivers , one  on each side for Y axis  moves.

Similar to another App. "Grbl Controller 3.61"   but this application is the best so far, but still not working.   Using an Original Uno with a Shield.

[image]

jahnj0584, 2017-08-27:
All you do is clone your y step + y dir pins to the other driver. I've done
it on three machines.

Sunnysky, 2017-08-28:
Yes I just discovered it but no invert on direction 

Cheers,
Tony

pelrun, 2017-08-28:
Reversing a stepper motor just needs you to swap two of it's wires going to the driver (either of the two pairs is fine).

winder, 2017-08-28:
@Sunnysky Cloning the signals and wiring one of the Y axis motors in reverse is the usual way people do this. I think the feature you found in Grbl Controller is for a custom GRBL fork which isn't really what you want for dual-Y motors.

closed 2017-08-28 by winder

Sunnysky, 2017-08-28:
Thankyou for your fast response. that worked

Warmest regards,
Tony

Sunnysky, 2017-08-29:
Actually what I did was rotate connector on Shield header so that both 
phases were reversed.

  I really liked the new look of GGcode Viewer 2.0 and it has amazing 
improvements but I really miss the # parameter definitions and found the 
UX a bit taxing for me  to use.   there are some fine features and I 
admire your skills.

But I had to switch to another new discovery for me  Grbl Panel which is 
stable and complete but without graphic simulation which I did not find 
useful because our scale was over 1 meter square.

Cheers,
Tony

On 28-Aug-17 4:56:28 PM, "Dean A. \"Tony\" Stewart"