winder/Universal-G-Code-Sender #726: Settings appear to be misreported by $G at startup

groberts22, 2017-08-30:
I am using the 8/30 nightly build.  When I first connect, the values reported when the $G command is issued do not match the settings (In the attached screenshot, $13=1 yet $G reports G21 when the machine is first connected.  Once the machine is homed, the value changes automatically and then reports consistently.) I never noticed this before, and find it confusing.  Is this a GRBL characteristic?

[image]

winder, 2017-08-30:
Yes, this is a GRBL thing. `$13` is the `Report in inches` flag. It isn't related to the gcode state, it is literally the units being used in status reports.

closed 2017-08-30 by winder

groberts22, 2017-08-31:
Thanks.  I think I misrepresented the issue in my previous comment.  What is actually happening is this.  I include a G20 command at the end of the GRBL config.h file so that the machine is initialized to accept coordinate values in inches.  This is as suggested in the comments to the file.  As near as I can tell, this command is not issued until a homing cycle is run.  UGS reports the active gcode state when the machine is first connected, which is before the g20 command is sent.  When the homing command is subsequently issued, the g20 command is sent and the $G command shows the gcode state I am expecting to see.  Anyway, it probably wasn't worth mentioning in the first place, but at least I now know (I think) what is happening.